RobotLocomotion/drake #11065: symbolic_py broken on mac (after update?)

RussTedrake, 2019-03-28:
```
ERROR: /Users/russt/drake/bindings/pydrake/BUILD.bazel:205:1: Couldn't build file bindings/pydrake/_objs/symbolic.so/symbolic_py.pic.o: C++ compilation of rule '//bindings/pydrake:symbolic.so' failed (Exit 1) wrapped_clang failed: error executing command external/local_config_cc/wrapped_clang '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG ... (remaining 509 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
bindings/pydrake/symbolic_py.cc:256:21: error: explicitly assigning value of variable of type 'const pybind11::detail::self_t' to itself [-Werror,-Wself-assign-overloaded]
      .def(py::self -= py::self)
           ~~~~~~~~ ^  ~~~~~~~~
bindings/pydrake/symbolic_py.cc:274:21: error: explicitly assigning value of variable of type 'const pybind11::detail::self_t' to itself [-Werror,-Wself-assign-overloaded]
      .def(py::self /= py::self)
           ~~~~~~~~ ^  ~~~~~~~~
2 errors generated.
```

```
31-33-136:drake russt$ clang --version
Apple LLVM version 10.0.1 (clang-1001.0.46.3)
Target: x86_64-apple-darwin18.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

soonho-tri, 2019-03-28:
I think it's resolved by https://github.com/RobotLocomotion/drake/pull/11060. Could you check if you can reproduce it with the latest `master`?

soonho-tri, 2019-03-28:
Possible duplicate of #11037

RussTedrake, 2019-03-28:
i suspect you're right.  i had rebased this morning, but probably missed this commit.  the laptop is taking forever to recompile, and i have to run... so I will assume it is good.  apologies!

closed 2019-03-28 by RussTedrake

soonho-tri, 2019-03-28:
Not a problem, please let me know if the problem doesn't go away.