RobotLocomotion/drake #11072: [geometry] Re-examine `GeometrySet` API

SeanCurtis-TRI, 2019-03-28:
## Background

The `GeometrySet` was originally used to communicate geometries to the "collision filtering" mechanism. It was intended to be a accumulator to express collision filters concisely. So, it's functionality is centered around its *construction* in an effort to make it as easy as possible to create a set (as exhibited by):
  - Multiple overloads to allow for specifying geometries via:
      - individual ids
      - initializer list of ids
      - arbitrary containers of ids (e.g., std::set, std::vector, etc.).
  - Shorthand for geometries fixed to a frame (i.e., specify a frame to implicitly include all geometries fixed to that frame). And corresponding overloads.

The consumption of the contents was intended to be an internal operation -- and in fact, mapping `FrameId` to `GeometryId` requires (more or less) a `SceneGraph` instance to perform the mapping.

The set had an interface that was introduce to facilitate testing (`geometries()` and `frames()`) that returned the contained geometries and frames. 

## The problem statement

Apparently, the usage of `GeometryState` has grown beyond this initial intention. There were two historical issues that arose:
  - Merging `GeometrySet` instances was problematic. The presence of `geometries()` implied the ability to instantiate one set from another, but frames were accidentally omitted. This led to #10417 which allowed automatic merging of sets and deprecated the direct access to the geometries and frames (preserving their existence for testing).
  - The presence of the `geometries()` method has led to it being used to iterate through the geometries. For a `GeometrySet` instance of unknown origin, this does not guarantee coverage of all geometries "in" the set (as indicated by the previous issue). For a controlled instance, in which only `GeometryId` values have been added, this would be valid. But the API is deprecated for good reason.

The problem is that this class seems insufficiently set-like -- it *suggests* it's a set of geometries, but breaks down in important ways (iterability, set operations, etc.)

## Possible solutions

1. [One suggestion](https://github.com/RobotLocomotion/drake/pull/11067#issuecomment-477741503) arose in the discussion of another PR. Essentially, eliminate the class from the public API and simply use some stl container in its place.
2. [Alternatively](https://github.com/RobotLocomotion/drake/issues/10990#issuecomment-476297459), we can extend the API to provide a more complete set of...well...set-compatible operations. (Although this alternative doesn't address the ability to iterate through the geometries.)
3. Suggestions?

## Discussion

The proposed solutions have pros and cons.

1. The advantage of creating `GeometrySet` is that the convenience constructors can be isolated in one place. These constructors facilitate testing and allow users to collect ids in sets, vectors, initializer lits, whatever they like. If we eliminate this class, then every function that would consume a set of geometry would either a) explode the API or b) lose the flexibility (although not *all* flexibility; it could still be expressed, for example, in terms of generic containers).
2. Eliminating `GeometrySet` would require its replacement to be expressed solely in terms of `GeometryId`. That means while it might be worth thinking about "this body shouldn't collide with *that* body", which, under the current regime, can be specified by "this frame doesn't collide with that frame", now two sets of geometries would need to be created to perform that operation. And it would have to be done dynamically to account for the possibility that the representation maintained on disk might change the geometries associated with a link.
3. In order to iterate through the geometry ids implied by a `GeometrySet`, we would either have to eliminate the `FrameId` short-hand or make it a `Context` dependent-operation. This doesn't seem strictly worthwhile. If a user has some `GeometryId`s they want to iterate through, they can maintain them in any stl data structure that is convenient to them. Then, when it comes to the `SceneGraph` API, the call would be `SceneGraph::some_func(GeometrySet(my_ids));`. This could be facilitated by an interface that *shares* the data with `my_ids` to avoid copying if that's a concern. Or simply copy it for simplicity.
4. Other thoughts?

SeanCurtis-TRI, 2019-03-28:
/cc @calderpg-tri as a major stakeholder in this conversation.

rpoyner-tri, 2026-11-11:
Reassigning to @SeanCurtis-TRI, since this is a geometry design question in your area rather than something I'm working on. Please unassign yourself or push back if that's wrong.