RobotLocomotion/drake #11087: Parse collision filter from SDF files

SeanCurtis-TRI, 2019-03-29:
PR #8232 had allowed collision filter groups to be defined in an sdf file. This functionality was supported by the RBT implementation. As we've moved to MPB/SG, this feature has not yet made the transition.

Rather than blindly aping that specification, we should examine it and determine if it needs tweaking before we implement. This issue will serve as the basis to clearly define the feature we want.

SeanCurtis-TRI, 2019-03-29:
cc @calderpg-tri @EricCousineau-TRI

EricCousineau-TRI, 2019-03-29:
Marginally relates (but may not require) #9754

huihuaTRI, 2019-12-11:
Can I bump up the priority of this issue since the FLT uses the collision filter tag heavily? It seems now we have a clear user request.

SeanCurtis-TRI, 2019-12-18:
#12495 is a direct translation of the original RBT functionality. We'll consider that it closes this PR and if "rethinking" is truly required, we'll open a new PR.

huihuaTRI, 2019-12-18:
Would this issue needs a separate PR other than [#12495](https://github.com/RobotLocomotion/drake/pull/12495) since that one is only for URDFs?

SeanCurtis-TRI, 2019-12-18:
That's a good point. We should leave this open, noting that the implementation is only half in place.

SeanCurtis-TRI, 2019-12-18:
Or should #12495 also add it to SDF at the same time?  :D

huihuaTRI, 2019-12-18:
I am actually looking into that already. I will report back about the plan.

huihuaTRI, 2019-12-18:
Talked to Sean f2f, the plan is to push PR #12495 first and then work on the SDF part. The main thought pushes this decision is to remove the blocker on the FL stack which only uses URDFs.

PR #12495 (2019-12-20): Parse the collision filter group for urdfs
This is a verbatim translation of the original RBT parsing functionality mapped into the new MBP/SG collision filtering API, but preserving the semantics of the filter group declarations found in URDF.

resolves #11087

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12495)
closed 2019-12-20 by huihuaTRI