RobotLocomotion/drake #11094: SpatialInertia::IsPhysicallyValid() fails for reasonable point-mass models

RussTedrake, 2019-03-30:
I believe that the urdf at 

https://github.com/RussTedrake/underactuated/blob/master/src/double_pendulum/double_pendulum.urdf 

is a reasonable model, I think.  The spatial inertia for a link with a point mass, about the center of mass, is zero, and that should be ok.  Do you agree?

sherm1, 2019-03-30:
Yes, I do agree. Even completely _massless_ (and inertialess) should be acceptable -- might cause trouble at runtime but might be fine as long as some outboard bodies have mass.

@amcastro-tri is there something wrong with the IsPhysicallyValid() test?

RussTedrake, 2019-03-30:
ironically, it means that i cannot port my very first example (Example 1.1) from underactuated to mbp.
http://underactuated.csail.mit.edu/underactuated.html?chapter=intro

RussTedrake, 2019-03-30:
it's not a mystery, IsPhysicallyValid() asks for the spatial inertia to be strictly positive definite.

amcastro-tri, 2019-04-01:
`SpatialInertia` does allow point masses (zero rotational inertia and non-zero mass) and even zero masses.
The problem in that file are not the point masses but the inertia for `<link>` "base". The inertia for that link does not satisfy the triangle inequality since you have `Ixx=0.01`, `Iyy=Izz=0`. Therefore the inequality `Iyy + Izz >= Ixx` does not hold (0 + 0 < 0.01). If the base is a "thin-rod" like object, then having two of the principal moments different from zero and the third one equal to zero is allowed. Also, you can make all moments equal to zero and that is also allowed (I tried locally).

This makes me think we need better error messages.

sherm1, 2019-04-01:
Thanks, Alejandro! @RussTedrake, can we close this issue since #11109 is open?

RussTedrake, 2019-04-01:
I see.  I think that the reason that this urdf is valid for RigidBodyPlant but invalid for MultibodyPlant is because of the deferred welding of the joints?  In RigidBodyPlant land, the base inertia's meant nothing... because i immediately specified the floating joint = weld.  In MultibodyPlant i do not have that option?

RussTedrake, 2019-04-01:
Perhaps I'm wrong... if all zeros for the base works, then I think I should have no complaint.   Will close.

closed 2019-04-01 by RussTedrake