RobotLocomotion/drake #11096: Receiving AttributeError 'NoneType' object has no '_Bound'

tklinterg, 2019-03-31:
Hi, 

I'm trying to run the test file for pydrake related to scalar conversion but get an AttributeError stating that 'NoneType' object has no attribute '_Bound'.

I've tried the Xenial distribution of pydrake for Unix.

Does someone happen to know where the issue might be ?

Thanks in advance.

EricCousineau-TRI, 2019-04-03:
Howdy @tklinterg!
Can I ask if you have self-contained example that can reproduce this error?

If you haven't seen it already, and if it helps any, here's an example scalar-converting system:
https://github.com/RobotLocomotion/drake/blob/9a63cc75ed6d338b8d10396f3ae7a6ef52637ff3/bindings/pydrake/systems/test/scalar_conversion_test.py#L12-L27

I'm guessing it's something to do with this bit, but unfortunately I'm not sure what `TemplateMethod` might be triggering this...
https://github.com/RobotLocomotion/drake/blob/9a63cc75ed6d338b8d10396f3ae7a6ef52637ff3/bindings/pydrake/common/cpp_template.py#L362

EricCousineau-TRI, 2019-04-03:
Er, rereading your question, I'm guessing it may be that file specifically?

Can you post a [Gist](https://gist.github.com/) of the full stack trace of your exception?
And can you post the SHA of the version of Drake you're using? ([example info](https://drake.mit.edu/getting_help.html#helpful-information))

EricCousineau-TRI, 2019-04-22:
@tklinterg Can I ask if you can provide the above details?

If not, I may close this within the next day or so, as I am unaware how to reproduce issue.

In general, the error in `_Bound` should not be reachable as any added instantiation is checked to be non-`None`, so it leads me to believe that it may be something like Python version mismatching or a memory leak:
https://github.com/RobotLocomotion/drake/blob/9a63cc75ed6d338b8d10396f3ae7a6ef52637ff3/bindings/pydrake/common/cpp_template.py#L144

EricCousineau-TRI, 2019-04-26:
Closing this for now. Please do let me know if you run into this again!

closed 2019-04-26 by EricCousineau-TRI