RobotLocomotion/drake #11109: Better error messages when spatial inertias from file are not physical.

amcastro-tri, 2019-04-01:
In #11094 @RussTedrake was getting a "non-physically valid" spatial inertia error message when loading a model from URDF file. His initial thought was that the problem was with point masses since this model happened to have two point masses. However, further inspection revealed the problem was the "base" of the model had an inertia not satisfying the triangle inequality.

The error message however was not good enough to diagnose the problem without having to modify the URDF/dig in the code.

The proposed solution is to offer a better diagnosis message saying exactly what condition, for what body, was not satisfied.

tri-ltyyu, 2019-04-29:
If this issue is in progress, is there any update or next step? Thanks!

jwnimmer-tri, 2021-11-12:
This seems like a duplicate of #11109 at this point.

sherm1, 2021-11-12:
Agreed.

closed 2021-11-12 by sherm1