RobotLocomotion/drake #11110: Standardize naming for "count" methods to start with num_ rather than get_num_

sherm1, 2019-04-01:
Drake APIs have been inconsistent in naming simple "count" methods like `num_input_ports()` vs. `get_num_input_ports()`. PR #11078 make the System Framework consistently use `num_` for all such methods. Per discussion in that PR, we should consider adopting that standard Drake-wide, deprecating the old `get_` methods, and updating the style guide to reflect that convention.

jwnimmer-tri, 2019-04-01:
I am happy for the framework to be consistent, since the System and Context deal so frequently in terms of "how many foo's do I have".  I am not sure it's worth bothering with the rest of the tree.  I'd like to suggest closing this until it's a problem.

EricCousineau-TRI, 2019-04-01:
Er, just to follow-up: I'd be OK to close this as long as this rule is encoded somewhere for the Systems Framework (in a header file or something).

I guess I'd be OK if we chalk up to general consistency without an explicit policy, but it seems like explicit would be nice here?

jwnimmer-tri, 2019-04-01:
I'm content with either of those options.

jwnimmer-tri, 2020-04-21:
Over the past year, no re-occurrences of `get_num_foo` have regressed within the systems framework.  To me, that says the cargo-cult / implicit policy for spelling here is sufficiently robust.

@EricCousineau-TRI (or anyone) if you'd like to find somewhere in framework Doxygen to stash this advice then I have no objection, but in the meantime I don't think we need an issue reminding us.

Also, there seems to be no great pull (and I also personally disfavor) any broader project-wide (cppguide) naming rule.

Given all of that status, I'd rather focus on more useful issues instead, so I'm closing this.

closed 2020-04-21 by jwnimmer-tri