RobotLocomotion/drake #11117: scene_graph: De-hack (world, world) collision filtering

EricCousineau-TRI, 2019-04-01:
See discussion in #11116.

SeanCurtis-TRI, 2019-04-02:
Additional use case: 

A number of bodies are all fixed together into one *meta* body. We would like MBP to be able to report that to SG such that:
   - Only one pose needs to be reported for that rigid system and
   - collisions between the geometries associated with those bodies are automatically filtered.

Open question:
  - will we still want to be able to map from collision geometry to *original* body?

EricCousineau-TRI, 2019-04-02:
Er, haven't fully parsed, but initial knee-jerk reaction:

> Only one pose needs to be reported for that rigid system and [...]

I'm not sure if I'm totes on board with this one. If there is a good and consistent story on mapping between `Frame<double>` and SG's `FrameId`, then maybe, but I'd almost prefer that there be complete parity between the two, and prefer `FrameId` over `(FrameId, X_FG)` pairs (#10247).

siyuanfeng-tri, 2019-04-02:
> Only one pose needs to be reported for that rigid system and [...]
I don't like that idea either, keeping the geom <-> body association intact helps humans to understand the model better. 

The other argument is that suppose body 1 and 2 are joined by a revolute joint, and body 3 is welded to 2. We want to do collision checking between 1 and 3. If we collapse everything welded to 2 into 2, we can't do that anymore.

SeanCurtis-TRI, 2019-04-02:
I know what you mean, I was uneasy when I wrote it. That particular point is *purely* optimization. It reduces the amount of data needing to be prepared and sent over ports (and processing of same). But, clearly, it's not worth it.

In the same vein, the responsibility for automatic collision filtering between geometries in that fixed system will belong to MBP. (For the same reason that filtering collisions between adjacent bodies belongs to MBP.) SG knows nothing about either case.

EricCousineau-TRI, 2019-04-02:
So summary is, no action to take regarding welding + geometry associations, as we've stated that certain applications care about distinctions of bodies?
(Current functionality is sufficient; if applications want to lump the bodies, they must do so themselves?)

siyuanfeng-tri, 2019-04-02:
aye

SeanCurtis-TRI, 2019-04-03:
> So summary is, no action to take regarding welding + geometry associations

Not literally so, right? The change in #11116 is not going to be reverted, and we should still do it in a more legitimate way (although, this conversation has led me to wonder what "legitimate" means).

EricCousineau-TRI, 2019-04-04:
Ah, sorry, you're right! I meant no further action to take on neighboring link + welding associations.

Definitely still work to de-hack collision filtering for `(world, world)`.

rpoyner-tri, 2022-04-12:
I will claim that whatever objections of hackery existed three years ago have been superseded, by use and lack of complaints.

closed 2022-04-12 by rpoyner-tri