RobotLocomotion/drake #11120: CalcPointPairPenetrations should support AutoDiffXd (even if SceneGraph does not)

RussTedrake, 2019-04-02:
Per the discussion started here: https://stackoverflow.com/questions/55449337/this-method-only-supports-t-double-when-creating-lqr-for-multibodyplantdoub/55453269#55453269

We, of course, need CalcPointPairPenetrations to support AutoDiffXd, but per @avalenzu -- the full version is waiting on the gradients of the normals from SceneGraph.  (#10747 is the most relevant I can see right now).

But even now, CalcPointPairPenetrations could support AutoDiffXd, but and throw iff the number of penetrations pairs is non-zero.  That way people don't have to comment out the contact geometry (e.g. in their urdf/sdf) just to take a gradient for the cases where there isn't any active contact!  This was the way I had implemented it in RigidBodyPlant 
https://github.com/RobotLocomotion/drake/blob/master/attic/multibody/rigid_body_plant/compliant_contact_model.cc#L62

sherm1, 2019-04-02:
@SeanCurtis-TRI and/or @DamrongGuoy this sounds reasonable and easy to me?

DamrongGuoy, 2019-04-02:
I'll check details again later.  For now, I confirm the function to change is:
```
drake::geometry::QueryObject::ComputePointPairPenetration()
```
https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_query_object.html#a9ae8651607d21a9fb37348c83e6d65c4

Right now it returns `std::vector< PenetrationAsPointPair< double > >`, we want to change to `std::vector< PenetrationAsPointPair< T > >`  where `T` is the template parameter of QueryObject.

Do we want the same thing for `ComputeSignedDistancePairwiseClosestPoints()` and `ComputeSignedDistanceToPoint()`.

sherm1, 2019-04-02:
> Do we want the same thing for ComputeSignedDistancePairwiseClosestPoints() and ComputeSignedDistanceToPoint().

If possible, yes.

hongkai-dai, 2019-04-02:
BTW, I am working on returning the gradient of SceneGraph signed distance query w.r.t the relative pose of the two geometries. This issue is described in https://github.com/RobotLocomotion/drake/issues/10932. It computes the gradient of any quantities (distance, witness point, contact normal) w.r.t the relative pose. This is not as general as computing the gradient using AutoDiffXd, as AutoDiffXd allows us to compute the gradient w.r.t anything, not just w.r.t the relative pose. But I believe it is more efficient to just compute the gradient w.r.t the relative pose, for two reasons
1. AutoDiffXd is slower than a fixed size gradient, as the former involves dynamic memory allocation.
2. I can then compute the gradient of distance/witness points/ contact normal w.r.t `q` using MBP<double>::CalcJacobian function. Note that I don't need to use MBP<AutoDiffXd>.

There will be a train of PRs to address #10932, the first PR #11086 has been merged.

RussTedrake, 2019-04-02:
@hongkai-dai -- great.  I would be awesome to slowly add support for the contacts in these functions of MBP (throw for less and less cases) as the support becomes available!

RussTedrake, 2019-04-02:
I think that the only acceptable answer is to return accurate gradients or THROW.

I think that without taking any steps towards full geometry query support for AutoDiffXd, we can support a bunch of cases - namely those where the model has collision geometry but the current context has no collisions.  In this case, we call the double version of the geometry queries, see that there are no penetration, and return the accurate gradients.  If any contact pairs are returned, then we throw.

of course i want to take more steps down that road, but this is my current request.

SeanCurtis-TRI, 2019-04-02:
Thanks @RussTedrake. That does a great job of defining the shape of the immediate PR.

The guiding principle being: only throw if there is an answer that gets derivatives that can't be guaranteed. (It might also be a reasonable justification for introducing a `bool HasCollisions()` query.)

rcory, 2019-04-03:
Fyi, see PR #11143 for the no collision case.

tri-ltyyu, 2019-04-12:
See PR #11208.

SeanCurtis-TRI, 2019-04-29:
Current shape-to-shape infrastructure close to PR (single test remaining).

SeanCurtis-TRI, 2019-05-06:
PR #11739 under review; two lgtms with some cosmetic touches remaining.

DamrongGuoy, 2019-05-06:
I think @SeanCurtis-TRI meant #11379 (Autodiff support for the pairwise signed distance query).

SeanCurtis-TRI, 2019-05-13:
Between PR #11143 and #11379 the core of this issue is closed.

There is some interesting follow up to do. This has been captured in new issues #11454 and #11455. This issue is closed in favor of those.

closed 2019-05-13 by SeanCurtis-TRI