RobotLocomotion/drake #11138: pydrake systems that derive from LeafSystem spew deprecation warnings from _NAME

RussTedrake, 2019-04-03:
Pretty sure I've purged all bad usages from my code.  But the tests (including the homework we will post today) still spew deprecation warnings.

The simplest example I have is
https://github.com/RussTedrake/underactuated/blob/master/src/van_der_pol/particles.py
(which also uses) https://github.com/RussTedrake/underactuated/blob/master/src/underactuated/pyplot_visualizer.py

but results in 
```
russt@Puget-179850-01:~/underactuated/src/van_der_pol$ python particles.py 
particles.py:81: DrakeDeprecationWarning: '_DoHasDirectFeedthrough' is deprecated and will be removed on or around 2019-06-15. Please call 'DoHasDirectFeedthrough' instead.
    Please see `help(pydrake.common.deprecation)` for more information.
  diagram = builder.Build()
/home/russt/underactuated/src/underactuated/pyplot_visualizer.py:53: DrakeDeprecationWarning: '_DoPublish' is deprecated and will be removed on or around 2019-06-15. Please call 'DoPublish' instead.
    Please see `help(pydrake.common.deprecation)` for more information.
  LeafSystem.DoPublish(self, context, event)
/usr/local/lib/python2.7/dist-packages/matplotlib/backend_bases.py:2453: MatplotlibDeprecationWarning: Using default event loop until function specific to this GUI is implemented
  warnings.warn(str, mplDeprecation)
particles.py:87: DrakeDeprecationWarning: '_DoCalcTimeDerivatives' is deprecated and will be removed on or around 2019-06-15. Please call 'DoCalcTimeDerivatives' instead.
    Please see `help(pydrake.common.deprecation)` for more information.
  simulator.StepTo(20)
```

cc @weigao95 , @YunzhuLi

EricCousineau-TRI, 2019-04-03:
For now, can this be worked around by adding a warning filter?
At which point, this priority could be lowered to `medium`, at least until May, at which time the warning should be removed, and this should be fixed?

EricCousineau-TRI, 2019-04-03:
Investigating with drake@54d2c41 and `underactuated@5fedce4`. Will post updates in this comment...

PR #11150 (2019-04-04): py systems: Demote `_{NAME}` deprecation to trigger on function call
If used as a property deprecation, it can triggered by `py::get_overload`.
This was missed because of some bug in either
(a) `pybind11` (our fork or the real thing) or (b) CPython.

Resolves #11138

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11150)
commit 533f8d21 (2019-04-04): py systems: Demote `_{NAME}` deprecation to trigger on function call

If used as a property deprecation, it can triggered by `py::get_overload`.
This was missed because of some bug in either
(a) `pybind11` (our fork or the real thing) or (b) CPython.

Resolves #11138
closed 2019-04-04 by EricCousineau-TRI