RobotLocomotion/drake #11148: System should provide a way to make all cache entries up to date

sherm1, 2019-04-03:
When using the "freeze context" feature requested in #10442 and implemented in PR #11145, any accessed cache entries must already be up to date. A user who knows which cache entries are required while the context is frozen can invoke `Eval()` methods on those entries to bring them up to date before freezing. However, in cases where performance prior to freezing isn't critical, it would be easier just to ask the System to update everything using something like `system.UpdateAllCacheEntries()`.

cc @calderpg-tri

Implementation notes:
- updates can leak out of the current subsystem via its input ports
- inter-cache entry dependencies must obeyed when modifying a cache entry directly -- that can make invalidations leak out of the current subsystem via its output ports

rpoyner-tri, 2026-11-11:
Unassigning myself; I'm not actively working on this.

sherm1, 2026-11-11:
7 1/2 years later we haven't done anything with this; I think we can safely close it as "won't do".

closed 2026-11-11 by jwnimmer-tri