RobotLocomotion/drake #11154: Compute contact surface from pressure equilibrium isosurface

tri-ltyyu, 2019-04-04:
This is a placeholder issue for computing contact surface which is needed as Spiral I of Hydroelastic force model implementation

DamrongGuoy, 2019-06-01:
I started working on the contact surface between a compliant object and a rigid object.

Step 1.  I am writing a code to compute the intersecting polygon between a triangle and a tetrahedron.  The triangle is from the surface of a rigid object, and the tetrahedron is from the volume of a compliant object. I'll try to PR early next week.

Step 2. Assemble all polygons from Step1 to define the contact surface and compute its pressure distribution.  Hopefully I can start in the middle of next week.

I'll focus on this issue in the next two weeks before my vacation.

CC: @sherm1 , @edrumwri , @SeanCurtis-TRI , @amcastro-tri

DamrongGuoy, 2019-06-15:
I have been working on PR #11612 to solve the soft-rigid case.  It passes the feature review now but is still under platform review.  Since I'll be on vacation, @SeanCurtis-TRI will polish it and see it through the merge.

I'll work on the soft-soft case when I'm back.

DamrongGuoy, 2020-09-30:
We already had rigid-soft case working. I'll close this issue now. In the future, we can create another issue for soft-soft case.

closed 2020-09-30 by DamrongGuoy