RobotLocomotion/drake #11167: Operational transpose and inverse for RotationMatrix and RigidTransform

SeanCurtis-TRI, 2019-04-04:
Currently, calling [`RotationMatrix::transpose()`](https://github.com/RobotLocomotion/drake/blob/master/math/rotation_matrix.h#L322), [`RotationMatrix::inverse()`](https://github.com/RobotLocomotion/drake/blob/master/math/rotation_matrix.h#L315) or [`RigidTransform::inverse()`](https://github.com/RobotLocomotion/drake/blob/master/math/rigid_transform.h#L287-L288) create copies of the original data. Eigen returns [`Transpose`](https://eigen.tuxfamily.org/dox/classEigen_1_1DenseBase.html#ac8952c19644a4ac7e41bea45c19b909c) for transpose operations to avoid such duplication.

It would be good to have similar operational objects that get returned for transpose and inverse to eliminate needless copying to memory.  Something akin to:

`RotationTranspose RotationMatrix::transpose()`
`RotationTranspose RotationMatrix::inverse()`
`RigidTransformInverse RigidTransform::inverse()`

in which they share data with the originating structure and overload the appropriate mathematical operators.

jwnimmer-tri, 2019-04-04:
> Eigen returns Transpose ...

Eigen _also_ has the `eval()` / `auto` fiasco: https://eigen.tuxfamily.org/dox/TopicPitfalls.html.  I'm not sure I want Drake to share that prize.

Do you have a benchmark that shows copying at fault for slowness?

sherm1, 2019-04-04:
Agree we should know we're gaining something useful before doing this. But limited use of an inverse type doesn't have to introduce eval() ugliness (Simbody has exactly this, used extensively with no trouble.)

However, for max performance the methods transpose() and inverse() should return references -- they should be just casts:
```c++
const RotationMatrixInverse& RotationMatrix::inverse() const;
const RigidTransformInverse& RigidTransform::inverse() const;
```

sherm1, 2019-04-04:
BTW there is a fuzzier reason for doing this too -- conscientious scientific programmers get upset seeing unnecessary copying going on in their inner loops. That might incline people to be reluctant to convert from Eigen's Isometry to our RigidTransform, even if it doesn't actually matter much. I'd like to be able to say we don't do any copying either, and in fact returning references would make our inverse operations slightly better than Eigen's.

EricCousineau-TRI, 2019-04-08:
> [...] I'm not sure I want Drake to share that prize.

+1

If people are concerned for `RotationMatrix`, they can do `rot.matrix().transpose()` at present if they really want to pay the expression template cost.
And yeah, for `RigidTransform`, would definitely love to have this motivated by a concrete benchmark.

All that being said, agreed on the priority tag.

sherm1, 2019-04-08:
@SeanCurtis-TRI suggested a much less-painful compromise: just add a few more methods to RigidTransform and RotationMatrix to support common inverse operations, e.g. 
`p_BQ = X_AB.ApplyInverseToPoint(p_AQ);` or whatever is a good name. There aren't many of those.

jwnimmer-tri, 2019-04-08:
Until we have a benchmark that shows that we care, I don't think that we should add redundant and confusing method sugar.

sherm1, 2019-04-08:
> Until we have a benchmark that shows that we care ...

While that is generally a good philosophy, it is not the only one applicable to low-level tools. Eigen, for example, did many optimizations "on spec" and I would argue that that is partially responsible for its wide adoption. Personally I'm glad to know Eigen isn't making unnecessary copies of my vectors and matrices, even if I can't prove it actually matters in my current applications. I don't want to see programmers avoiding RigidTransform for performance reasons, real or imaginary. So I don't think it would be a terrible thing to add some special handling for fast operations with inverse matrices on spec. In any case we will need to at least draft them in order to run any performance comparisons. And I'm skeptical that we'll get dispositive experiments. Likely outcomes are: (1) we design a test that does lots of extra copying so it "proves" we need the new methods, or (2) we run with existing code where a lot else is going on and it "proves" we don't need them. Neither test actually proves anything.

jwnimmer-tri, 2019-04-08:
Okay, let me try it another way:  I'll retract:
> Until we have a benchmark that shows that we care, I don't think that we should add redundant and confusing method sugar.

... and instead just say:

**I don't think that we should add redundant and confusing method sugar.**

A serious yoda readability problem `p_BQ = X_AB.ApplyInverseToPoint(p_AQ);` has.  Adding such a monster should require an exhaustion argument in support of it -- that the performance is bad, and there is literally no other way to solve the problem.  That means benchmarking.

jwnimmer-tri, 2024-03-16:
Well, the `RigidTransform::InvertAndCompose` and `RotationMatrix::InvertAndCompose` horses already escaped the barn, so I think this issue is fixed.

closed 2024-03-16 by jwnimmer-tri