RobotLocomotion/drake #11170: Hydroelastic model Spiral II

tri-ltyyu, 2019-04-04:
This is to track the second phase of hydroelastic pressure field modeling. 

The expected outcome fo this phase is the ability to simulate two geometrically simple contacting objects (e.g., boxes, cylinders, spheres) with automatic meshing and such that the existing simple gripper test runs within 5% of the speed of our existing sphere-based approach using some integrator.

edrumwri, 2020-01-02:
Reassigning based on my departure. Please feel free to reassign to someone else.

SeanCurtis-TRI, 2020-05-14:
@amcastro-tri @sherm1 

We might want to consider closing this PR. We've achieved almost all of it -- we know we can simulate contact between plenty of different primitive types that are automatically meshed. The [rolling sphere example](https://github.com/RobotLocomotion/drake/tree/master/examples/multibody/rolling_sphere) is concrete evidence of sphere-half space and sphere-box contact (with all sorts of rigid/soft combinations).

There is *one* missing aspect of the issue description that is missing: "within 5% of the speed" of point contact. We're not even close. Hydroelastic should be less than factor of 1.05X more expensive. On my laptop, it's actually 15X. I think that is such a big thing it's probably worth closing this issue which contains both code artifacts and performance for one that just tracks our performance efforts.

sherm1, 2020-05-14:
Agreed. Would be good to replace this with an Epic issue capturing this performance goal. Then we can point to individual issues for (a) taking bigger steps (e.g. Alejandro's super-solvers), and (b) making the steps faster (e.g. Damrong's mesh speedups).

amcastro-tri, 2020-05-18:
I agree. I think this issue is outdated now.

sherm1, 2020-05-18:
Extracted and refined the performance requirement into #13337.

closed 2020-05-18 by sherm1