RobotLocomotion/drake #11179: Algebraic loop errors can be misleading

rcory, 2019-04-05:
For example, I have a system that declares an (abstract) input port that is left unconnected and unprocessed. When building the diagram containing this system, I get a detailed error indicating an algebraic loop, e.g.
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Algebraic loop detected in DiagramBuilder:
  drake/systems/controllers/PidController@0000559826636ab8:Out(0) depends on
  drake/systems/Adder@0000559826644cc0:In(0) depends on
  drake/systems/Adder@0000559826644cc0:Out(0) depends on
  drake/examples/pendulum/PendulumPlant@000055982663e2b0:In(0) depends on
  drake/examples/pendulum/PendulumPlant@000055982663e2b0:Out(0) depends on
  drake/systems/controllers/PidController@0000559826636ab8:In(0)
```
However, if that "dummy" port is removed, no errors are reported.

There are a few problems with this error message (from a user perspective). For one, I know that my system doesn't contain any algebraic loops (and this error message seems overly aggressive in asserting that I'm wrong). The other problem is, it is specifically listing the port sequence where it thinks the algebraic loop exists, which is again, not correct. In fact, in my case, the dummy port in question exists on index 2 (and not index 0 as is being reported in the error).

It seems that the reason this exception is being thrown is because my system does not support `ToSymbolic()`, which would allow the symbolic analyzer to rule out some cases. Since my system cannot convert to symbolic, the algebraic loop detector reports the first port indices it finds (i.e., 0) as the culprits.

Even if my system did support symbolic, I suspect that the analyzer still wouldn't be able to make a reasonable conclusion on any abstract ports.

A better error message would have been to omit any detailed port index listing (which is confusing to the user if no loops actually exist) and perhaps instead report something like:
```Cannot covert system to symbolic and therefore cannot properly detect algebraic loops. Consider overriding DoHasDirectFeedThrough directly.```

There were four Drake developers circled around my desk for about 15 minutes trying to figure out the cause of the error, which reinforces the need for better error reporting here.

jwnimmer-tri, 2019-04-05:
Could you post your example code that creates the Diagram?  I suspect part of the port index confusion comes from how, e.g., PidcontrollerSystem works, and not the systems framework or diagram itself.  Part of the fix might be repairing PID control error messages, not just the framework.

rcory, 2019-04-05:
You can take a look at [this](https://github.com/rcory/drake/blob/98a12cb8f3566569a27a65052565b1637224151f/examples/pendulum/trajectory_optimization_common.h#L68-L125) chunk of code which runs a simulation. If I remove [this](https://github.com/rcory/drake/blob/98a12cb8f3566569a27a65052565b1637224151f/examples/pendulum/pendulum_plant.h#L72) override for `HasDirectFeedThrough()` I get the algebraic loop error. 

Alternatively removing the dummy input port, by deleting [this](https://github.com/rcory/drake/blob/98a12cb8f3566569a27a65052565b1637224151f/examples/pendulum/pendulum_plant.cc#L59) line and [this](https://github.com/rcory/drake/blob/98a12cb8f3566569a27a65052565b1637224151f/examples/pendulum/pendulum_plant.cc#L224) line, also gets rid of the algebraic loop error. Mind you, that port is doing nothing at the moment (although in this case, it is connected to a `geometry::QueryObject` output port, but disconnecting doesn't change the error).

jwnimmer-tri, 2019-04-08:
The indices that the error message refers to are correct.  When `RegisterGeometry` has not been called yet, then `PendulumPlant` supports symbolic form, so `EXPECT_FALSE(plant.HasAnyDirectFeedthrough());` passes (as per the unit test).  Once you call `RegisterGeometry` then the plant no longer supports symbolic form, so `u0` feeds through to `y0` and we have a (conservatively declared, but spurious) algebraic loop.  The unit test fails to re-check the plant's shape when visualization has been enabled.  Indeed, there is _no_ direct unit test support for the visualization.  Fundamentally the approach in #9363 to adding visualization pollutes and complicates systems' dynamics.  Visualization should be a separate system that gets cascaded, full stop.

I'll work on improving the error message report, so that it's easier for the reader to identify by inspection which edge in the loop is the false dependency.

edrumwri, 2019-04-08:
Aside from messages about the topology, one message that would have been helpful to us as we were trying to sort through the logic was that the system was not ToSymbolic-able. The HasDirectFeedthrough() method describes the logic of the algebraic loop detector fairly well, but that logic is dependent upon whether the system is convertible to SymbolicExpression. I personally find that getting the right magic incantation for scalar conversions involves lots of cursing.

Rick thought that he had the right incantation, but after we had him insert a System.ToSymbolic() to check, we learned otherwise. Once we found that it threw, the debugging was easier.

Another message that would be helpful would express the conservative nature of the algebraic loop detector. The algebraic loop detector should say that it _thinks_ it found a loop, but it might be wrong and then it should point the user to documentation for resolving this problem. The documentation for HasDirectFeedthrough(.) is a start, but a recipe (kind of like what we have in the system scalar conversion page) would be nice.

edrumwri, 2019-04-08:
> Yes, I'll see if I can add more details along those lines to the error
> message.
>
> However:
>
> Rick thought that he had the right incantation, ...
>
> This is where Beyonce-driven development should kick in... "If you liked
> it you should have put a test on it." Tests are not only for the benefit of
> CI, nor only reviewers or other developers, they are for the benefit of
> *you*, the developer! The first thing I did when I took Rick's repro was
> turn it into a unit test. That should happen *before* any debugging
> occurs. It should happen concurrently with development, not just added
> after-the-fact to placate the code reviewers.
>
> Trying to guess whether some giant Diagram is doing what you want is
> difficult, even with good error messages. If you want to rely on the fact
> that PendulumPlant does not have direct feedthrough, then write that down
> in a unit test. Debugging it at the application layer is a waste of time
> (as evidenced by the "four Drake developers circled around my desk for
> about 15 minutes").

Not disagreeing with you but I still think just outputting "hey, this
system is not symbolic convertible" would be nice.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

jwnimmer-tri, 2019-04-08:
WIP so far in #11189:
```
Declared algebraic loop detected in DiagramBuilder:
  InputPort[0] (u0) of System ::adder (Adder<double>) is direct-feedthrough to
  OutputPort[0] (sum) of System ::adder (Adder<double>) is connected to
  InputPort[0] (u0) of System ::pass (PassThrough<double>) is direct-feedthrough to
  OutputPort[0] (y0) of System ::pass (PassThrough<double>) is connected to
  InputPort[0] (u0) of System ::adder (Adder<double>)
The most likely cause of this error is that one of the 'is direct-feedthrough to' lines
above indicates that a System's output port is declared to depend on one of its input
ports, but such a dependency is spurious. In that case, remove the spurious dependency
as shown in the API documentation for LeafSystem::DoHasDirectFeedthrough.
```

See also #11194 for a related enhancement.

amcastro-tri, 2019-04-09:
nice! I was able to draw the block schematic of this diagram just from this message!

jwnimmer-tri, 2019-04-18:
Progress:
- #11212 has merged, so that error messages are somewhat better.
- #11244 has merged, so that the pendulum doesn't have feedthrough bugs in the first place.

Plans:
- #11201 in review to make sparsity declarations more ergonomic.
- Write a new Doxygen section explaining how to manage feedthrough:
   - Prefer to express dependencies at DeclareOutputPort-time.
   - Explain how and when symbolic inspection can also help.
   - Disprefer DoHasDirectFeedthrough overrides; maybe even deprecate it.
   - Update all related APIs to cite the new section.
   - Update the exception message to cite the new section.
- Replicate #11244's fix for the other analogous //examples/...

Problems:
- None

jwnimmer-tri, 2019-04-22:
Progress:
- #11201 has merged (makes sparsity declarations more ergonomic).

Plans:
 - Use #11297 to deprecate DoHasDirectFeedthrough, and as part of write the new Doxygen section as enumerated upthread.

jwnimmer-tri, 2019-05-01:
Progress:
- #11298 has merged, closing #11297.

Plans:
- Write the new Doxygen section as enumerated upthread.

jwnimmer-tri, 2019-05-20:
Progress:

 * None

Plans:

  * Write the new Doxygen section as enumerated upthread.

PR #11684 (2019-06-24): framework: Improve direct-feedthrough documentation
Closes #11179.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11684)
closed 2019-06-24 by jwnimmer-tri