RobotLocomotion/drake #11184: LeafSystem::DeclareWitnessFunction => LeafSystem::MakeWitnessFunction

RussTedrake, 2019-04-07:
Just lost a bunch of time because I forgot that "DeclareWitnessFunction" does not actually declare anything.  It makes a witness function, but that witness function is irrelevant unless it is later returned by `DoGetWitnessFunctions`.  I recommend we change the name.

I've almost got python bindings working (took way too long).  They are [here](https://github.com/RobotLocomotion/drake/compare/master...RussTedrake:py_witness?expand=1).  Let's decide if we should change the name before I introduce the bindings.

EricCousineau-TRI, 2019-04-08:
I'd say yes. This breaks consistency with the other `Declare*` methods, and we should definitely fix that soon.

Actually, reading more into this, will post in the Python PR with another alternative.

EricCousineau-TRI, 2019-04-08:
Really, what'd prefer, if we had `LeafSystem::MakeFunction((&MySystem::calc*)(...))`, then code like this:
```
auto witness = this->DeclareWitnessFunction("witness", kCrossesZero, &MySys::stuff);
```
becomes:
```
auto witness = std::make_unique<WitnessFunction>("witness", kCrossesZero, MakeFunction(&MySys::stuff));
```

I think in lieu of sugar to save ~10 characters, we could have modularized the API.

jwnimmer-tri, 2019-04-08:
> Just lost a bunch of time because I forgot that "DeclareWitnessFunction" does not actually declare anything. It makes a witness function, but that witness function is irrelevant unless it is later returned by DoGetWitnessFunctions. I recommend we change the name.

Maybe we should band-aid some names / docs / bindings for now, but isn't the best long-term answer to have DeclareWitnessFunction actually (fully) declare the thing?  Having to override DoGet... is not cohesive.  Maybe the function should-self express whether it should be active, or maybe a separate function passed into Declare should do that.  But having _two_ methods in play is difficult API.

edrumwri, 2019-04-08:
@jwnimmer-tri on the surface that would seem to be correct, but it's actually not a good idea because one wants the set of active witness functions to be able to be dependent upon the current state.

jwnimmer-tri, 2019-04-08:
My expectation was that the "is active" predicate should be given `const Context&` just like `DoGetWitnessFunctions` is already given.  (Though, the State seems like it should be enough.)

edrumwri, 2019-04-08:
`MakeWitnessFunction(.)` would be good.

edrumwri, 2019-04-08:
> My expectation was that the "is active" predicate should be given `const Context&` just like `DoGetWitnessFunctions` is already given. (Though, the State seems like it should be enough.)

Sorry, which "is active" predicate?

jwnimmer-tri, 2019-04-08:
I mean either (or both) of:

>  ... maybe the function should-self express whether it should be active, ...

A new method
```c++
boolean<T> WitnessFunction::is_active(const Context<T>&) const { ... }
```

> ... maybe a separate function passed into Declare should do that ...

A new (required) option at declare-time (possibly with kAlways as a default?):
```patch
--- a/examples/rimless_wheel/rimless_wheel.cc
+++ b/examples/rimless_wheel/rimless_wheel.cc
@@ -35,6 +35,7 @@ RimlessWheel<T>::RimlessWheel()
   step_backward_ = this->DeclareWitnessFunction(
       "step backward",
       systems::WitnessFunctionDirection::kPositiveThenNonPositive,
+      &RimlessWheel::get_double_support,
       &RimlessWheel::StepBackwardGuard, &RimlessWheel::StepBackwardReset);
 
   step_forward_ = this->DeclareWitnessFunction(
```

edrumwri, 2019-04-08:
This API was designed to avoid querying a quadratic number of witness functions every AdvanceTo(.) call, as you might have with a simulation with n interacting geometries. A broad phase method in that case could drastically pare down the number of narrow phase calls (the witness functions).

`is_active(.)` would suffer in this case.

jwnimmer-tri, 2019-04-08:
I don't mind a "compute the active set in a single shot" API available.  That should not prevent us from having a Declare method that keeps the simple cases simple.  Having the feature broken into two methods (one of which is a constructor, and the other is an override) is _so difficult to use as to be impossible_.  We are being mean to users on purpose.  There a ways to solve the single-shot problem and still have good APIs.

My point here is not to dish out a solution.  My point is to underline that two methods is an _awful_ API design, and should be considered a usability defect.  I can dish out solutions when we come to fixing it.

jwnimmer-tri, 2019-04-08:
> MakeWitnessFunction(.) would be good.

FYI Russ planning on doing this, in the meantime.

edrumwri, 2019-04-08:
> I don't mind a "compute the active set in a single shot" API being available. That should not prevent us from having a Declare method that keeps the simple cases simple. Having the feature broken into two methods (one of which is a constructor, and the other is an override) is _so difficult to use as to be impossible_. We are being mean to users on purpose. There a ways to solve the single-shot problem and still have good APIs.

Agreed- usability is wanting at present. Interfaces that throw Russ for a loop should be improved.

jwnimmer-tri, 2019-04-08:
Maybe for now `Declare...` could stay around (not renamed nor deprecated), but default to always enabling the function for all states.  Users who want to have it disabled in certain states could override DoGet, but nobody else would have to.

edrumwri, 2019-04-08:
That could work though that would also mean that `DeclareWitness` would need to store the witness functions in the class (which I didn't do as it has its own potential O(n^2) problems).  Just brainstormed with @sherm1 for a bit about this. It would seem that we could solve some of these O(n^2) problems by returning a VectorX rather than a scalar from the witness functions. I had considered that originally but believed that returning n scalar witness functions was equivalent to returning an n-dimensional vector from a single witness. The framework does not require the VectorX to have a fixed size, so we could:

(a) Declare the witness functions up front
(b) Store them in the System
(c) Return them all by default in DoGetWitnessFunctions(.) which the user could override to return just a subset, if desired.

Thoughts?

RussTedrake, 2019-04-08:
#11195 is sufficient to resolve the original intent of this PR.  @jwnimmer-tri , @edrumwri -- please rename if you want this thread to stay active, or close (pending #11195)