RobotLocomotion/drake #11185: Extra periodic event is triggered after a (separate) WitnessFunction triggers

RussTedrake, 2019-04-07:
The reproduction is in the `custom_test.py` `test_leaf_system_overrides` method in the PR that I will link momentarily.

Before adding a witness function (w/ an unrestricted update event), the periodic update triggered exactly once -- as expected -- it's period is 1 second and we are simulating until 0.99.  But after introducing a witness function which trips at 0.5 seconds, I get a new periodic update event that occurs at the 0.5 seconds.  This was unexpected.  I believe it is an error?

cc @sherm1

edrumwri, 2019-04-08:
@RussTedrake which PR is that?

edrumwri, 2019-04-08:
Ugh, debugging through the python bindings :)  Thanks for making it easy for me to reproduce.

Will check this out tomorrow.

RussTedrake, 2019-04-08:
i agree.  debugging the core (header-only) libraries through the python bindings is no fun.  even adding a print statement requires recompiling all of `libdrake.so` to run a simple test.  I _believe_ a minimal c++ reproduction would be an equivalent LeafSystem with just the periodic update and the guard/reset witness.

jwnimmer-tri, 2019-04-08:
From platform review pr PR #11186: it adds a comment `TODO(edrumwri): Uncomment this pending resolution of #11185`.  The PR that fixes this issue should be sure to go and clean up that comment.

PR #11196 (2019-04-16): Fixes witness functions causing periodic triggers to trigger early. 
Resolves #11185.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11196)
closed 2019-04-16 by edrumwri