RobotLocomotion/drake #11187: DoGetWitnessFunctions(context, witnesses) => witnesses = DoGetWitnessFunctions(context)

RussTedrake, 2019-04-07:
Write now `Systems` have
```
  /// Derived classes can override this method to provide witness functions
  /// active for the given state. The default implementation does nothing. On
  /// entry to this function, the context will have already been validated and
  /// the vector of witness functions will have been validated to be both empty
  /// and non-null.
  virtual void DoGetWitnessFunctions(const Context<T>&,
      std::vector<const WitnessFunction<T>*>*) const
```
I've always found this signature strange (especially the comment guaranteeing that it will be empty).  It also doesn't appear to be avoiding any memory allocations in the simulator.  And in #11186, I found out the hard way that this is a particularly nasty -- at least for those with my level of experience with pybind -- signature for python.

I propose we change it to 
```
  virtual std::vector<const WitnessFunction<T>*> DoGetWitnessFunctions(const Context<T>&) const
```
which is also what I ended up binding for python (so far). 

Would be a low priority cleanup, except that it may block our decision on getting any witness function support into python.

sherm1, 2019-04-08:
@edrumwri, please take a look.

edrumwri, 2019-04-08:
While memory allocations are not missing from Simulator, they're not baked into the interfaces. Unless the implementer happens to be knowledgable about return value optimization (rare IMO), they might not know what they need to do to avoid this memory allocation. The interface I provided was built with that logic.

I am sympathetic to the unusual warning (FYI, that's there to prevent that vector from attempting to be used as anything but an output argument) and any and all pybind woes. I'd vote for the more performant version, but I do like the simplicity of the interface you proposed too.

RussTedrake, 2019-04-08:
I'm not actually sure that the nominal c++ user does any better with the current interface?  Even if I have a pre-allocated list of witness pointers that I want to always return, don't I need to copy them into the passed `witnesses` vector pointer, (e.g with `witnesses.insert(witnesses.end(), myvector.begin(), myvector.end())`?

RussTedrake, 2019-04-08:
But witnesses almost cannot be reused, given the contract that you always pass in an empty vector?  Can you show me a pattern that actually accomplishes this?

RussTedrake, 2019-04-08:
Ah, you’re saying the vector has reserved memory, even though the size is zero?

edrumwri, 2019-04-08:
You got it.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA