RobotLocomotion/drake #11191: Support and recommend  "named vector views" in python.

RussTedrake, 2019-04-08:
Formerly #11172, which resulted in a discussion on `namedtuple` that ended in 
"Turns out namedtuples are immutable. So they don't handle my intended use cases. There are some alternatives, but they are not in the core libraries
https://bitbucket.org/ericvsmith/namedlist (and my quick impression is that namedlist will do a copy of the data, not a view)."

Eric agreed to look at the link above or write our own.

EricCousineau-TRI, 2019-04-08:
Some comments on `namedlist`:
* It's written by another Eric - too many Eric's! :P
* This generates a type based on `tuple`; our immediate usage is for `np.array`; I may tinker with a shorter version that focuses on this.
* The implementation is very complex for what I think could be simple, if we directly subclass `list` / `np.ndarray` / whatever. (Mayhaps it's done this way for efficiency, but I don't think I want to pay high maintenance cost for what may be a low cost operation in the first place.)

Will tinker with this in a bit.

EricCousineau-TRI, 2019-04-08:
@RussTedrake Here's the generalized `namedview` prototype:
https://github.com/EricCousineau-TRI/repro/commit/9d6c349092b563d24cea255602814e9071f9e942#diff-6a3353602f69e14a6b57ed718bfb394cR78

How does this look to you?

RussTedrake, 2019-04-08:
Looks great to my eyes.  Can take it for a spin tonight

RussTedrake, 2019-04-10:
It's great.  But I do have a few feature requests/issues.  Given that I have 
- [ ] `view.wrong_field = 2` fails silently.  please throw an error.  (lost some time to this!)
- [ ] `print(view)` is just a list of numbers.  would be nice to print the same as `namedtuple`
- [ ] would love to do `SetFromVector(view)`. currently I have to do `SetFromVector(view[:])`

RussTedrake, 2019-04-10:
in case it's helpful, here is a sample usage:

```
import numpy as np
%matplotlib notebook
import matplotlib.pyplot as plt
from collections import namedtuple

from pydrake.systems.analysis import Simulator
from pydrake.systems.framework import (BasicVector, DiagramBuilder, LeafSystem, PublishEvent,
                                       UnrestrictedUpdateEvent, WitnessFunctionDirection)
from pydrake.systems.primitives import ConstantVectorSource, LogOutput

SLIPState = namedview('SLIPState', ['x','z','length', 'theta', 'xdot', 'zdot', 'lengthdot', 'thetadot'])

class SpringLoadedInvertedPendulum(LeafSystem):
    def __init__(self):
        LeafSystem.__init__(self)

        self.DeclareVectorInputPort("touchdown_angle", BasicVector(1))
        self.DeclareContinuousState(BasicVector(np.zeros(8)), 4, 4, 0) 
        
        self.DeclareVectorOutputPort("state", BasicVector(8), self.CopyStateOut)
        
        self.mass = 80.  # kg
        self.rest_length = 1.  # m
        self.gravity = 9.8  # m/s^2
        self.stiffness = 14000. # spring coefficient
        
        self.last_apex = None  # placeholder for writing return map result.
        
        self.touchdown_witness = self.DeclareWitnessFunction(
            "touchdown", WitnessFunctionDirection.kPositiveThenNonPositive, 
            self.foot_height, UnrestrictedUpdateEvent(self.touchdown))
        self.takeoff_witness = self.DeclareWitnessFunction(
            "takeoff", WitnessFunctionDirection.kNegativeThenNonNegative,
            self.foot_height, UnrestrictedUpdateEvent(self.takeoff))
        self.apex_witness = self.DeclareWitnessFunction(
            "apex", WitnessFunctionDirection.kPositiveThenNonPositive,
            self.apex, PublishEvent(self.publish_apex))
                                    
    def foot_height(self, context):
        s = SLIPState(context.get_continuous_state_vector().CopyToVector())
        return s.z - self.rest_length*np.cos(s.theta)

    def touchdown(self, context, event, state):
        s = SLIPState(context.get_mutable_continuous_state_vector().CopyToVector())
    
        # Update lengthdot and thetadot to match xdot and ydot, using 
        # x = -length*sin(theta), y = length*cos(theta)
        s.length = self.rest_length
        s.lengthdot = (s.x*s.xdot + s.z*s.zdot)/self.rest_length
        s.thetadot=-(s.xdot+s.lengthdot*np.sin(s.theta))/(self.rest_length*np.cos(s.theta))
        state.get_mutable_continuous_state().get_mutable_vector().SetFromVector(s[:])

    def takeoff(self, context, event, state):
        s = SLIPState(context.get_mutable_continuous_state_vector().CopyToVector())

        # Update theta to commanded leg angle.
        s.theta = self.EvalVectorInput(context, 0).GetAtIndex(0)
        s.thetadot = 0
        s.length = self.rest_length
        s.lengthdot = 0
        
        state.get_mutable_continuous_state().get_mutable_vector().SetFromVector(s[:])
        
    def apex(self, context):
        return context.get_continuous_state_vector().GetAtIndex(5)  # zdot
    
    def publish_apex(self, context, event):
        # TODO(russt): provide an option to terminate here instead, pending #4447.
        if self.last_apex is None:
            s = SLIPState(context.get_mutable_continuous_state_vector().CopyToVector())
            self.last_apex = s.z
        
    def CopyStateOut(self, context, output):
        x = context.get_continuous_state_vector().CopyToVector()
        y = output.SetFromVector(x)

    def DoGetWitnessFunctions(self, context):
        return [self.touchdown_witness, self.takeoff_witness, self.apex_witness]
        
    def DoCalcTimeDerivatives(self, context, derivatives):
        s = SLIPState(context.get_continuous_state_vector().CopyToVector())
        sdot = SLIPState(np.zeros(8));
        sdot[0:4] = s[4:8]

        if (self.foot_height(context) < 0):
            # then we're in "stance"
            sdot.lengthdot = self.stiffness/self.mass*(self.rest_length-s.length) \
                          + s.length*s.thetadot**2 \
                          - self.gravity*np.cos(s.theta)
            sdot.thetadot = self.gravity/s.length*np.sin(s.theta) - 2*s.lengthdot*s.thetadot/s.length

            # x = -length*sin(theta), y = length*cos(theta) =>
            sdot.xdot = - sdot.lengthdot*np.sin(s.theta) - 2*s.lengthdot*s.thetadot*np.cos(s.theta) \
                          + s.length*s.thetadot**2*np.sin(s.theta) - s.length*sdot.thetadot*np.cos(s.theta)
            sdot.zdot = sdot.lengthdot*np.cos(s.theta) - 2*s.lengthdot*s.thetadot*np.sin(s.theta) \
                          - s.length*sdot.thetadot*np.sin(s.theta) - s.length*s.thetadot**2*np.cos(s.theta)
        else:
            sdot.xdot = 0
            sdot.zdot = -self.gravity
            sdot.lengthdot = 0
            sdot.thetadot = 0
            
        derivatives.get_mutable_vector().SetFromVector(sdot[:])
        

touchdown_angle = 0 #0.1
s = SLIPState(np.zeros(8))
s.z = 1.5
s.theta = touchdown_angle
s.length = 1
#s.xdot = 0.7

# parameters from Geyer05, figure 2.4
#touchdown_angle = np.deg2rad(60)
#tildeEs = 1.61
#tildek = 10.8

builder = DiagramBuilder()
plant = builder.AddSystem(SpringLoadedInvertedPendulum())
#viz = builder.AddSystem(SLIPVisualizer())
#builder.Connect(plant.get_output_port(0), viz.get_input_port(0))
log = LogOutput(plant.get_output_port(0), builder)

command = builder.AddSystem(ConstantVectorSource([touchdown_angle]))
builder.Connect(command.get_output_port(0), plant.get_input_port(0))

diagram = builder.Build()
simulator = Simulator(diagram)
context = simulator.get_mutable_context()
context.SetAccuracy(1e-5)

context.get_mutable_continuous_state_vector().SetFromVector(s[:])

simulator.set_target_realtime_rate(1.0)
simulator.StepTo(5.0)

print("apex: " + str(plant.last_apex))

t = log.sample_times()
x = log.data()
fig, ax = plt.subplots(2)
ax[0].plot(t, x[1, :])
ax[1].plot(x[0, :], x[1, :])
plt.show()
```

EricCousineau-TRI, 2019-04-12:
Fixed 3/4 of the checkboxes:
https://github.com/EricCousineau-TRI/repro/commit/fdb67a80f7d4fcafa3c1a96adae5c160d8fb9184

Not sure what's causing the last one about being able to pass vectors in. Will see if I can check that briefly now.

EricCousineau-TRI, 2019-04-12:
This should fix the NumPy conversion, and check the last (3rd) box:
https://github.com/EricCousineau-TRI/repro/commit/a60daf899e9726daf2ca1259bb80ad2c7c9b3e3f
Ultimately, this interacts with this bit of `pybind11`, which is effectively reflected by `np.asarray`:
https://github.com/pybind/pybind11/blob/9bb3313162c0b856125e481ceece9d8faa567716/include/pybind11/numpy.h#L797-L798

RussTedrake, 2019-04-15:
worked with the newest version this morning.  all great.  can we land it?

PR #11248 (2019-04-15): py containers: Add `namedview`
Resolves #11191

Well, it resolves the "support" part. Recommend is a bit nebulous still; can I ask for recommendations on this part?

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11248)
closed 2019-04-15 by RussTedrake