RobotLocomotion/drake #11192: CDash: Investigate SQS FIFO Queues

BetsyMcPhail, 2019-04-08:
**Background:**
CDash asynchronously parses its submissions. When a build client (typically jenkins) submits results to CDash, they go on an Amazon SQS queue. We use a third party message queue (MQ) abstraction library called bernard to interact with SQS.

**Description:**
On the afternoon of 2019/04/04, we noticed that CDash had fallen about 90 minutes behind in parsing its submissions. We tracked this down to CDash failing to delete an existing build, which in turn caused it to fail to process and delete the message on the front of the queue. Once the problematic build was removed manually, processing resumed as normal.

We also noticed that CDash was attempting to parse the same message once every 10 minutes. Although all four consumers were running and ready to process messages, no other messages were deleted from the queue.

BetsyMcPhail, 2019-04-08:
https://aws.amazon.com/blogs/developer/how-the-amazon-sqs-fifo-api-works/

> For the VisibilityTimeout period, the messages you’ve received are hidden from other callers and, because this is a FIFO queue, other callers are partially blocked from reading those messages to preserve FIFO ordering.  “Partially blocked” has to do with MessageGroupId.  If all the messages have the same one, then the whole queue is blocked; if there are several different MessageGroupIds, only the MessageGroupIds in the batch you receive are blocked.

The `VisibilityTimeout` is set to 10 minutes

BetsyMcPhail, 2019-04-08:
Bernard does support SQS FIFO Queues: https://github.com/bernardphp/bernard/pull/318

However, the `MessageGroupId` is always the same: https://github.com/bernardphp/bernard/blob/3ff88921bea87186d1baffc93708dd7aee4e963e/src/Driver/Sqs/Driver.php#L173

tri-ltyyu, 2019-04-11:
This has been fixed by Zack. Closing

closed 2019-04-11 by tri-ltyyu