RobotLocomotion/drake #11193: Update multibody call sites to use RigidTransform (part 3).

amcastro-tri, 2019-04-08:
Continuation of the work started in #10959 and #11064.

As with #11064, feel free to assign your name and address as many of these you'd like.

A brute force search of `Isometry3` within `drake/multibody` and `pydrake/multibody` revels the following occurrences of `Isometry3` across 25 files (number counts how many times in that file):

- [ ] bindings/pydrake/multibody/plant_py.cc: 5
- [ ] bindings/pydrake/multibody/test/inverse_kinematics_test.py: 1
- [ ] bindings/pydrake/multibody/test/plant_test.py: 2
- [ ] bindings/pydrake/multibody/tree_py.cc: 3
- [ ] multibody/benchmarks/kuka_iiwa_robot/drake_kuka_iiwa_robot.h: 1
- [ ] multibody/benchmarks/kuka_iiwa_robot/make_kuka_iiwa_model.cc: 1
- [ ] multibody/inverse_kinematics/test/inverse_kinematics_test_utilities.cc: 2
- [ ] multibody/inverse_kinematics/test/inverse_kinematics_test_utilities.h: 2
- [ ] multibody/inverse_kinematics/test/minimum_distance_constraint_test.cc: 2
- [ ] multibody/multibody_doxygen.h: 2
- [ ] multibody/parsing/detail_urdf_geometry.cc: 5
- [ ] multibody/parsing/test/detail_scene_graph_test.cc: 5
- [ ] multibody/parsing/test/detail_sdf_parser_test.cc: 9
- [ ] multibody/parsing/test/detail_tinyxml_test.cc: 4
- [ ] multibody/parsing/test/detail_urdf_geometry_test.cc: 3
- [ ] multibody/plant/multibody_plant.cc: 1
- [ ] multibody/plant/multibody_plant.h: 25
- [ ] multibody/plant/test/frame_kinematics_test.cc: 7
- [ ] multibody/plant/test/kuka_iiwa_model_tests.h: 1
- [ ] multibody/plant/test/multibody_plant_jacobians_test.cc: 1
- [ ] multibody/plant/test/multibody_plant_test.cc: 11
- [ ] multibody/test_utilities/spatial_kinematics.h: 5
- [ ] multibody/tree/fixed_offset_frame.h: 8
- [ ] multibody/tree/multibody_tree.cc: 2
- [ ] multibody/tree/weld_joint.h: 2

amcastro-tri, 2019-04-29:
This has gone a big stale. I believe @mitiguy is working on some of this, could you confirm @mitiguy? I'll ask for more volunteers in the dynamics channel.

mitiguy, 2019-04-30:
Yes @amcastro-tri  I am on this.  I got side-tracked slightly with numerical integrator test.

mitiguy, 2019-05-06:
Monday update: Work to restart after onsite.

mitiguy, 2019-05-20:
Progress update Monday May 20th -- background information only
All progress last week was on updating the Jacobian interface (Issue in progress #10155),
none on this issue.
Merged  PR#11414 12 days ago which checked off the following boxes in this issue:
multibody/benchmarks/kuka_iiwa_robot/drake_kuka_iiwa_robot.h: 1 @mitiguy
multibody/benchmarks/kuka_iiwa_robot/make_kuka_iiwa_model.cc: 1 @mitiguy
multibody/plant/test/frame_kinematics_test.cc: 7 @mitiguy
multibody/plant/test/kuka_iiwa_model_tests.h: 1 @mitiguy
multibody/plant/test/multibody_plant_jacobians_test.cc: 1 @mitiguy
multibody/plant/test/multibody_plant_test.cc: 11 @mitiguy

amcastro-tri, 2019-05-20:
#11414 landed. Two `drake/multibody` bullets remaining, @mitiguy is it a good assumption that you'll address them.

@SeanCurtis-TRI, regarding the last bullet mentioning an additional `GeometryInstance` ctor, is that a sensible solution?

mitiguy, 2019-05-20:
Yes - I'll address them this week.

amcastro-tri, 2019-05-22:
Excellent! we are only one bullet away from resolving this one!
Thanks to everyone who contributed.

jwnimmer-tri, 2019-05-22:
FWIW I don't think we should do anything about the last checkbox -- I think we should close this now.

SceneGraph is all-in on Isometry3; we shouldn't try to start shaving that yak under the umbrella of this issue.

mitiguy, 2019-05-28:
Closed this issue and opened issue #11549 which deals specifically with SceneGraph.

closed 2019-05-28 by mitiguy