RobotLocomotion/drake #11194: Default input-output feedthrough should use prerequisites_of_calc

jwnimmer-tri, 2019-04-08:
When a System declares that an OutputPort `calc` function does not use some inputs, the author should not _also_ have to override `DoHasDirectFeedthrough` to tell the framework that same thing again.  The LeafSystem should know the answer already and reply accordingly.

sherm1, 2019-04-09:
Ideally yes! The default for the prerequisites is very conservative though (as are those for HasDirectFeedthrough) so it's not clear this will be less misleading. Lots of false positives at the moment.

jwnimmer-tri, 2019-04-09:
Indeed, the immediate effect would be small.  However, it seems _much better_ instruct users to add a prereqs argument to their Declare when they have feedthrough problems, than to override DoGetHasVerbNounThingForAllPortsIntMap or whatever it's called.  Doing it at declare-time is more cohesive, and it will also make caching (much?) more effective.

PR #11201 (2019-04-19): framework: Use LeafOutputPort declared feedthrough
Closes #11194.  Relates #11179.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11201)
closed 2019-04-19 by jwnimmer-tri