RobotLocomotion/drake #11219: Create unsigned-16 label image

SeanCurtis-TRI, 2019-04-10:
Currently, the only label image type that is defined uses signed, 16-bit integers to represent the labels. However, label values are limited to just the positive range. (Reducing the possible unique labels by half). This came up in a [recent PR](https://reviewable.io/reviews/robotlocomotion/drake/11042#-Lbjge8L9xwDftvkRYQI:-LbyKIXTD7cN5t8xUrZe:b-3tbiln).

So, we can simply *add* the image type. However, it needs to be supported all the way through the lcm pipeline so that such images, if produced, can be reasonably consumed. Once the image exists, the `RenderLabel` class can be tied to *that* image type, and the `RenderEngine` class can produce it.

jwnimmer-tri, 2019-04-10:
I would actually advocate for a signed 32-bit label image type as higher priority.  If someone needs the last bit of label palette, chances are they need more than just one bit anyway.  And 32 bits is more likely to support per-instance labels instead of per-kind labels for many use cases.

SeanCurtis-TRI, 2019-07-31:
Closing as "won't do". The point that this only serves people who definitely need more than 32K labels but no more than 64K labels is probably a small user group. If and when such a use-case comes up, it can be re-opened.

jwnimmer-tri, 2024-07-30:
Re-opening.  We don't care about the extra 1 bit, but we _do_ care that images are unsigned numbers!  No image format can save signed numbers, so we save label images on disk as 16U monochrome, leading to user confusion.