RobotLocomotion/drake #11229: Some MultibodyTree documentation should come out of internal::

sherm1, 2019-04-11:
Per @EricCousineau-TRI's Slack comment:

> On the topic of `internal`, the only acute pain point of entirely scrubbing this is that I will lose access to the `Mobilizer` documentation; I **really** like being able to point a `drake.mit.edu` URL with that doc, 'cause it's awesome.
> e.g. I like the "Mathematical Description of a Mobilizer" section:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1internal_1_1_mobilizer.html
> If we find a way to hoist that out of `internal`, great. Otherwise, I'd like that doc to stay alive somehow?

My suggestion would be to extract the useful documentation into a Doxygen Module and place it in the [Multibody Kinematics and Dynamics group](https://drake.mit.edu/doxygen_cxx/group__multibody.html) on Drake's doxygen front page.

cc @amcastro-tri @EricCousineau-TRI

EricCousineau-TRI, 2019-04-11:
FTR This was discussed in the context of hiding Doxygen symbols that are in the `internal` namespace:
https://drakedevelopers.slack.com/archives/C2NN588UA/p1554929438013000

amcastro-tri, 2019-04-12:
I like the idea of a separate Doxygen module. I think we'll be able to do this cleanup more effectively once we sort out the split between a "physical representation" and a "modeling representation". Lots of progress in #10068.

closed 2024-11-02 by jwnimmer-tri