RobotLocomotion/drake #11234: Policy for better consistency for C++ and Python argument names

mitiguy, 2019-04-11:
## Background

In submitting PR#11198, there were various opinions on the names of arguments to a constructor and two overloads of operator* (multiply).  This name issue may be bigger than this PR.  Up to now, the choice of an argument name seems to have competing influences from:
* A desire to adhere to standard notation  https://drake.mit.edu/doxygen_cxx/group__multibody__notation.html
    - This leads to parameters named with the notation, e.g., `void foo(const Vector3& p_AoBo_A)` instead of `void foo(const Vector3& position)`.
* Agreement between C++ and Python documentation
    - Creating Python documentation by simply stealing C++ documentation __*requires*__ C++ parameter names to match Python keyword argument names.
* Smooth the road for Python keyword argument users, we would want the keyword to be consistent across multiple distinct, but otherwise equivalent classes.  E.g., :
      - `RigidTransform(const RigidTransform& pose)`
      - `RigidTransform(const Translation3& pose)`
In both cases, the parameter is a "pose", despite the fact they are different classes. Leading to Python code like:

```py
pose1 = RigidTransform(...)
pose2 = Translation3(...)
X_AB = RigidTransform(pose = pose1)
X_BC = RigidTransform(pose = pose2)
```

## Problem statement

It is unclear, to what degree these factors need to be reconciled or what counts as a defect in code review.

## Proposal

1. We should document what is *required* to coordinate C++ and its corresponding Python code to facilitate Python users with clear demarcation of defect vs preference.
2. Ideally, the requirements will lead to a consistent pattern of keyword argument usage across classes in pyDrake (e.g., A "transform" should always have the keyword "pose", whether it is RigidTransform or Translation3).  This will increase the burden on C++ documentation to match the generic keyword name ("rotation") with notation suitable for discussion (`R_AB`); this is currently done to a limited extent, see e.g., [RigidTransform multiplication](https://github.com/RobotLocomotion/drake/blob/master/math/rigid_transform.h#L321-L327).

mitiguy, 2019-04-11:
cc @EricCousineau-TRI @jwnimmer-tri

jwnimmer-tri, 2019-04-11:
I am not sure there is any overarching policy which is unclear here.  You (as shown above) understand that C++ and Python argument names must match for documentation reasons, and that the argument name in Python is part of the signature.

Is the only confusion that in some cases overloads should prefer to use identical names, when there is no distinction?  That's not a Drake policy thing, that's a case-by-case API design decision.

jwnimmer-tri, 2019-04-11:
I suspect that dealing with the debate in the _context of actual code on the table_ is best, and we should discuss in the PR, and not this issue.

mitiguy, 2019-04-11:
The intent of PR# 11198 is to allow easy use of Eigen's Translation3 with RigidTransform in C++ (not Python).  It was my defect to indicate an intention to facilitate the use of Translation3 with Python.

The use of Translation3 in conjunction with RigidTransform in Python can either be done in a later PR or skipped.  As far as I know, there is no current use of Translation3 in conjunction with RigidTransform by Python users (the C++ code will merge when PR #11198 lands).

jwnimmer-tri, 2019-04-11:
That's all good.

I'm not sure how to resolve this issue, though?  I don't see that any policy is lacking, or that there is any additional work to be done.

EricCousineau-TRI, 2019-04-11:
I think that (1) is actionable by explicitly calling this out in the Python docs. Will draft up a PR for that.
It's present reading is a bit too implicit. At it's present incantation, I think the only things the docs do for args is state docstring naming rules, and kinda introduce (yet not require) `py::arg(...)`:
https://github.com/RobotLocomotion/drake/blob/984b03e751b220824881a043b97f2c2ea36eec36/bindings/pydrake/pydrake_doxygen.h#L129
https://github.com/RobotLocomotion/drake/blob/984b03e751b220824881a043b97f2c2ea36eec36/bindings/pydrake/pydrake_doxygen.h#L179

For (2), though, I think I'd agree with Jeremy that this is very case-by-case dependent, and I'm not sure we can do much of a generic policy other than "choose useful argument names".

But to make sure, was the discussion in the relevant PR related to renaming existing arguments -- in that there was too much inertia to rename it b/c Python slows down the change?
Or is it in introducing a new overload, and keeping consistency? (the two are kinda coupled, but I'd assume the pain point can mebbe be more tied to one...)

>  As far as I know, there is no current use of Translation3 in conjunction with RigidTransform by Python users (the C++ code will merge when PR #11198 lands).

FTR, there was only one existing usage of `Translation3` getting converted to/from Python, and that was actually an abuse; just filed #11236. I intend to shield Python from this type, as it doesn't seem actually help any workflows there.

jamiesnape, 2019-04-12:
Ideally, the documentation generation or some other linter would just scream when someone chose the wrong parameter name. I don't think that is beyond the realms of possibility going forward and we should try not to over-document.

EricCousineau-TRI, 2019-04-12:
Agreed, but #11238 (to address the simpler parts of documentation) is dirt-simple, and provides good value.

jwnimmer-tri, 2020-04-21:
I think a year without any progress supports my hypothesis above that there is no work remaining to be done, so I'm going to close this now.  If there is still any unclear naming policy, please point out what is lacking.

closed 2020-04-21 by jwnimmer-tri