RobotLocomotion/drake #11239: py ik: Bind and expose constraint classes

EricCousineau-TRI, 2019-04-12:
See Slack convo: https://drakedevelopers.slack.com/archives/C2CK4CWE7/p1555024805009600

I'm not sure if I'll get this into `master` anytime soon, but can slap something together as far as hacky WIPs go.

\cc @hongkai-dai @TobiaMarcucci

RussTedrake, 2019-04-14:
fwiw -- spent time on a work-around for this today (cleaning up the costs/constraints via std::function), which should provide something workable in python -- hopefully by tomorrow.

RussTedrake, 2019-04-14:
Just opened #11245 for discussion.

EricCousineau-TRI, 2019-08-06:
Per Slack convo with @manuelli, he may need this within the next month. Setting target date above.

EricCousineau-TRI, 2019-08-22:
Updated target date to next Thurs. @m-chaturvedi will take it from here - thanks!

m-chaturvedi, 2019-08-22:
@EricCousineau-TRI, I am not sure if I understand what we'd like to bind.  
If we'd like to modify [`inverse_kinematics_py.cc`](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/multibody/inverse_kinematics_py.cc), the class, [`InverseKinematics`](https://github.com/RobotLocomotion/drake/blob/master/multibody/inverse_kinematics/inverse_kinematics.h) is not templated (only has `double`'s).

If we'd like to modify [`ik_py.cc`](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/attic/solvers/ik_py.cc), it looks like most of the classes are [non-templated](https://github.com/RobotLocomotion/drake/blob/master/attic/multibody/rigid_body_constraint.h).

EricCousineau-TRI, 2019-08-22:
Yup, your edit is correct; this issue is about binding the constraint classes defined in `*_constraint.h`, so that it can be used for trajectory optimization (not just single-configuration IK).

PR #11959 (2019-08-30): py inverse_kinematics: Add constraint bindings
Resolves #11239

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11959)
closed 2019-08-30 by EricCousineau-TRI