RobotLocomotion/drake #11240: py mbp: Expose autodiff and symbolic scalar types

EricCousineau-TRI, 2019-04-12:
Relevant Slack discussion:
https://drakedevelopers.slack.com/archives/C2CK4CWE7/p1555082817001300?thread_ts=1555024805.009600&cid=C2CK4CWE7

> Thanks Russ and Michael!
> Russ: do you mean something like this?
> ```plant_sym = plant.ToSymbolic()
> q = prog.NewContinuousVariables(plant.num_positions())
> context = plant_sym.CreateDefaultContext()
> plant_sym.SetPositions(context, q)```
> If yes, here I get: `'System_[Expression]' object has no attribute 'SetPositions'`.

\cc @TobiaMarcucci @RussTedrake @hongkai-dai

EricCousineau-TRI, 2019-05-05:
Added checklist. I'm going to tackle the first few items to figure out the general procedure and do some much-needed cleanup.
After binding `.common.eigen_geometry`, may request help from @m-chaturvedi.

\cc @jamiesnape 

WIP Branch: https://github.com/RobotLocomotion/drake/compare/master...EricCousineau-TRI:issue/11240-wip

EricCousineau-TRI, 2019-05-15:
Per Slack convo, other suggestions for looking at scalar-type binding:
https://github.com/RobotLocomotion/drake/blob/2df723d/bindings/pydrake/systems/primitives_py.cc
https://github.com/RobotLocomotion/drake/blob/2df723d/bindings/pydrake/systems/framework_py_values.cc

This is a bit more advanced, so take it with a grain of salt:
https://github.com/RobotLocomotion/drake/blob/2df723d/bindings/pydrake/systems/framework_py_systems.cc

EricCousineau-TRI, 2019-05-23:
Primary diff with scalar type change for eigen geometry: https://github.com/RobotLocomotion/drake/commit/fc331ee68ad658c6ee26f394bef7326886e371f8

https://drake.mit.edu/python_bindings.html#differences-with-c-api

mpetersen94, 2019-05-31:
Is there an eta for autodiff support of RigidTransform/RotationMatrix in python?  I'm happy to help with the process if that would accelerate the timeline at all.

m-chaturvedi, 2019-05-31:
I'm working on it.  I need to write the relevant tests for it.  I need to know from Eric which all methods need to be exposed other than `RigidTransform` and `RotationMatrix`.

EricCousineau-TRI, 2019-06-03:
Er, added my thumbs-up without answering!
To clarify, it would be best to bind all methods that are scalar-type depended, unless there are issues.
If there gets to be too much boiler-plate for instantiating unittest test cases, then it should be simple enough to simplify that, helping with aliases and all that.

EricCousineau-TRI, 2019-06-04:
@mpetersen94 -- @m-chaturvedi just landed that PR. At present, the only thing not fully supported is `RigidTransform_[T].translation()` - but that should be resolved via #11575.

gizatt, 2019-06-05:
As I understand it this is blocking Python binding of `StaticEquilibriumConstraint` -- for timeline purposes, I'd love to have access to those bindings in a week or so, but I can adjust if that looks like it won't happen. Thanks!

EricCousineau-TRI, 2019-06-06:
Per f2f, we will target next Friday, June 14th. Thanks!

EricCousineau-TRI, 2019-06-13:
Per Slack convos, @m-chaturvedi is almost there; wrangling some issues with scalar type conversion with MBP+SG and matrix aliasing per #8116

EricCousineau-TRI, 2019-06-18:
Done! Thanks @m-chaturvedi!

EricCousineau-TRI, 2019-06-18:
(Even though #11575 is not yet resolved, can still fix that separately without needing this to be open.)

closed 2019-06-18 by EricCousineau-TRI