RobotLocomotion/drake #11242: [geometry] Signed distance: point-to-cylinder gives wrong gradient

SeanCurtis-TRI, 2019-04-12:
PR #11208 extended the `ComputeSignedDistanceToPoint()` for use with AutoDiff scalar. However, the code that provides correct answers when the scalar is double produces garbage derivatives of those same values. It needs to be re-expressed so that correct derivatives come out.

hjsuh94, 2024-08-09:
Are there any updates / efforts on this issue? I encountered that I cannot autodiff through point to cylinder signed distance queries - after a brief glimpse at the [code](https://github.com/RobotLocomotion/drake/blob/cb8b4a1c076407130e724c6a28a1596e9aaef95d/geometry/proximity/distance_to_point_callback.cc#L207), it is also not clear to me what exactly is breaking autodiff.

NatMatLogica, 2026-09-25:
@hjsuh94 asked in August 2024 whether anything had moved on this. Here is a current reproduction and a candidate cause.

I reproduced the point-to-cylinder derivative problem on Drake `99ba35c4bf821c84eeb2f6aee001a90163674a52` (19 March 2026), using the existing disabled cylinder cases in `distance_to_point_callback_test.cc`.

Enabling that test gives **8 failed assertions**: 6 are cap queries on the cylinder axis, and 2 are queries exactly on the circular rim. The primal geometry and signed-distance derivative checks pass; the failure is in the differentiated witness identity:

`p_WQ = p_WN + distance * grad_W`

For a cap interior, the analytic witness map is `p_GN = (x, y, ±lz/2)`, with Jacobian `diag(1, 1, 0)`. The largest reported matrix discrepancy is `1.610718310730327`, against a test tolerance of `8.881784197001252e-16`.

I also recorded the same `DistanceToPoint<T>::operator()(Cylinderd)` arithmetic through the real production translation unit with a reverse-mode trace. For a near-axis cap query, the reverse sweep reports:

`d(residual_y) / d(qy) = 1`

The direct `qy` dependency contributes `+1`, and nothing cancels it because the reconstructed nearest point carries no dependency on the original query `x/y`.

Changing only the near-axis reconstruction to retain the original query `x/y` restores that missing dependency in the nearest-point term. The same trace then shows a cancelling `qy` contribution of `-1`; the reported gradient changes from `[0, 1, 0]` to `[0, 0, 0]`, and `residual_y` goes from `2e-16` to `0`.

Reading the source rather than the trace, the reconstruction responsible appears to be the radial-basis branch at `distance_to_point_callback.cc:265-271`, used when rebuilding `p_GN` at `distance_to_point_callback.cc:288`. I would value a maintainer's view on whether that is the right place to fix it.

Retaining the original query `x/y` removes all six differentiable axis failures. The remaining two points lie exactly on the sharp rim, where the gradient is not unique; marking that fact in the existing tester makes the full 11-test target pass.

Would a regression based on the currently disabled cylinder cases, followed by the minimal reformulation above, be the preferred contribution route?

For disclosure: I used MatLogica AADC-NG's call-site tracer as an independent dependency-path diagnostic. It replays the callback arithmetic rather than instrumenting Drake as a whole. The Drake reproducer, analytic normal, and central finite difference do not require proprietary tooling; no AADC-NG source or binary is part of the proposed change.