RobotLocomotion/drake #11246: looking for libprotobuf.9 , but 10 was installed?

RussTedrake, 2019-04-15:
From a student trying to run pydrake from binary on Ubuntu 18.04

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/drake/lib/python2.7/site-packages/pydrake/__init__.py", line 32, in <module>
    from . import common
  File "/opt/drake/lib/python2.7/site-packages/pydrake/common/__init__.py", line 3, in <module>
    from ._module_py import *
ImportError: libprotobuf.so.9: cannot open shared object file: No such file or directory
```

"This is the error I am running into and have tried many different things to resolve itincluding installing from rpm and deb and also attempting apt-get (which couldn't find it). I have libprotobuf.so.10 I think (from checking with locate), but am not sure how to get 9."

jwnimmer-tri, 2019-04-15:
From last night's https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-bionic.tar.gz, I see:
```
$ readelf -d /opt/drake/lib/libdrake.so  | grep proto
 0x0000000000000001 (NEEDED)             Shared library: [libprotobuf.so.10]
```
That is, it does indeed want `so.10`.  Not sure why the user is getting `so.9` required.  Maybe the user downloaded the wrong URL?  Which URL did the user use, and on what date?

jamiesnape, 2019-04-15:
(The xenial packages are linked against .9)

RussTedrake, 2019-04-15:
Thanks guys.  I’ve asked on piazza to followup here.

kbrandes45, 2019-04-15:
Using the URL above, I was successfully. I must have grabbed the wrong URL. Thank you for the help!

closed 2019-04-15 by jamiesnape