RobotLocomotion/drake #11247: [geometry] Make semantics for geometry support in proximity queries consistent

SeanCurtis-TRI, 2019-04-15:
## Background

During the process of adding extended autodiff support for the proximity queries, a [core principle](https://github.com/RobotLocomotion/drake/issues/11120#issuecomment-479091723) was defined.  In short, if SceneGraph should only provide an answer if it's meaningful. In the context of AutoDiff, it means it shouldn't strip out derivatives.

The question is, what should it do *instead*? We have several cases so far:

  - For computing signed distance to a point, geometries for which no point-primitive test has been defined are simply ignored.
  - For AutoDiff'd signed distance queries, we can only report meaningful results for a subset of geometry pair types. (And likewise for penetration depth). 

## Problem statement

For these queries with limited domain, we need a consistent and reliable response to being given operands that lie outside of the supported domain. 

## Proposals

- Always silently ignore unsupported geometry. Document clearly what is supported and what will be ignored.
    - Pro: This will allow a fixed scene description that will gradually report more results as the supporting code fills in.
    - Con: A fixed scene that has unsupported geometry types that become supported geometry will end up leading to possibly changing behaviors that can confuse users.
- Throw in the face of unsupported operands.
    - Pro: No confusion about whether any particular geometry is supported.
    - Con: It'll require revisions of inputs as the domain of supported geometries increase.

sherm1, 2019-04-15:
Would be nice if we could issue warnings! Maybe in lieu we could have the limited queries report the number of geometries they ignored or something like that?

SeanCurtis-TRI, 2021-12-04:
Largely supplanted yes #10907.

closed 2021-12-04 by SeanCurtis-TRI