RobotLocomotion/drake #11252: MultibodyPlant should check Context validity in debug builds

avalenzu, 2019-04-16:
Passing a diagram context into `Frame::CalcPose()` results in the following assertion:
```
abort: Failure at bazel-out/k8-dbg/bin/external/drake/systems/framework/_virtual_includes/cache_and_dependency_tracker/drake/systems/framework/cache.h:702 in get_cache_entry_value():
condition 'has_cache_entry_value(index)' failed.
```
I spent a fair few minutes wondering what had gone wrong in my diagram before I realized that the issue was a context mismatch. If the assertion had come from a `MultibodyPlant` method this would have been more apparent. We could add
```
DRAKE_ASSERT_VOID(System<double>::CheckValidContext(context));
```
in user-facing non-systems-framework methods of `MultibodyPlant`.

amcastro-tri, 2019-04-16:
would you prefer ASSERT (only debug) or DEMAND?

sherm1, 2019-04-16:
Context-validating is potentially very expensive.

avalenzu, 2019-04-16:
I think ASSERT, definitely.

amcastro-tri, 2020-01-13:
See related issue #11801.

amcastro-tri, 2020-01-13:
Should this be an MBP feature or a `System` feature? @sherm1?

jwnimmer-tri, 2020-01-13:
The check proposed by #12564 should never be disarmed (i.e., not only in Debug builds).  For the "correctly-mated System vs Context" checking, System methods (like EvalTimeDerivatives) will embed the check.  For MbP-specific functions like GetFreeBodyPose, the MbP must call ValidateContext, because the framework can't do it.

sherm1, 2020-01-13:
Agreed. Eventually you would probably hit a framework method that would barf on the mismatch, but the delayed detection would be less helpful for users. (Would be better if the ValidateContext method would report the name of the failing API though.)

jwnimmer-tri, 2020-12-09:
I think that #14389 resolved this.

jwnimmer-tri, 2021-03-03:
While #14389 resolved the API directly on MbP, there are several other public classes on the MbT-related side that still lack calls to `ValidateContxt`, e.g. `Frame::CalcPoseInWorld` as noted in https://github.com/RobotLocomotion/drake/issues/12560#issuecomment-786366018.

amcastro-tri, 2021-03-04:
How expensive are those ValidateContext()? can they safely be placed on release builds? 
Also, I imagine we want them in MBP's public APIs, not MBT?

jwnimmer-tri, 2021-03-04:
Per [its API Doc](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system_base.html#a7622a0fb573704de7505804d57e66a79), calling `ValidateContext` is sufficiently fast for performance sensitive code.

PR #14389 already added the checks to all of the MbP methods.

What's still needed here is to add it to any of othe `multibody/tree/...` methods that take a `Context`, e.g., `Frame::CalcPoseInWorld` and others.

rpoyner-tri, 2021-06-03:
@amcastro-tri do you have any work-in-progress on this? If not, I think i have a patch that should do the job.

amcastro-tri, 2021-06-04:
Your patch is excellent. Thanks @rpoyner-tri!

jwnimmer-tri, 2021-08-10:
In #15489, I found that `MultibodyPlant::SetDefaultState` and `MultibodyPlant::SetRandomState` lack validation checks on the `State*` in/out pointer.  They call `CheckValidState`, but that function only does a basic "continuous xor discrete" sanity check, it doesn't inspect the SystemId tag.

PR #15129 (2021-06-07): multibody tree: Validate contexts at public methods
Closes #11252.

Since all of the context-taking methods in multibody/tree lead
eventually to MultiBodyTreeSystem, ensure that all of its public methods
validate the context.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15129)
commit de9c0fe5 (2021-06-07): multibody tree: Validate contexts at public methods (#15129)

Closes #11252.

Since all of the context-taking methods in multibody/tree lead
eventually to MultiBodyTreeSystem, ensure that all of its public methods
validate the context.
PR #15646 (2021-08-24): multibody: Properly validate State at public APIs
Closes: #11252

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15646)
commit 03173592 (2021-08-25): multibody: Properly validate State at public APIs (#15646)

Closes: #11252