RobotLocomotion/drake #11260: SceneGraph reports distance result for a specific pair of geometries

hongkai-dai, 2019-04-16:
Currently SceneGraph reports the distance results for every pair of geometries. If I just need the result between a specific pair (with given geometry ID), then SceneGraph does a lot of redundant computation.

SeanCurtis-TRI, 2019-04-17:
Related to #10990.

SeanCurtis-TRI, 2019-11-18:
It's not currently in Q4 OKRs. How much is it's absence affecting you?

hongkai-dai, 2019-11-18:
It's a performance improvement. So if you don't plan to work on it in Q4, that is fine. Do you think it will take a lot of time to add this feature?

SeanCurtis-TRI, 2019-11-18:
Well, as a one-off, I could probably knock this out quickly. As the original issue suggests, it relates to a more elaborate discussion in #10990 and I'd like to convince myself that doing a quick patch here doesn't conflict in any way with the bigger solution.

SeanCurtis-TRI, 2019-11-18:
How heavy is the performance hit? Do you have any sense of the size?

hongkai-dai, 2019-11-18:
I have a system with 10 geometries, so there are 55 different pairs. In many cases I only need 1 or 2 out of these 55 pairs.

SeanCurtis-TRI, 2019-11-18:
Right...and since this is *distance*, you're getting all pairs (i.e., no culling). So, what's a factor of 25X between friends?

DamrongGuoy, 2020-02-25:
@hongkai-dai , @SeanCurtis-TRI  I think we can close this issue now, right?

SeanCurtis-TRI, 2020-02-25:
Correct. This was closed by #12379.

closed 2020-02-25 by SeanCurtis-TRI