RobotLocomotion/drake #11275: Misleading exception on GetJointByName with wrong type

ggould-tri, 2019-04-18:
If you call `GetJointByName<T>(name, instance)` for a joint that is not `T` but appears in multiple instances, you will get the error "Joint ... appears in multiple model instances" (even though you specified an instance) which will send you down a long garden path of misplaced debugging.

The issue is that the generation of the *correct* error message ("Joint ... in model instance... is not of type ... but of type ...") calls `GetJointByName(name)`, not propagating `instance`, and so generating the right exception actually generates the wrong exception before it can do its work.

The fix might be as simple as 

```
--- a/multibody/tree/multibody_tree.h
+++ b/multibody/tree/multibody_tree.h
@@ -952,7 +952,7 @@ class MultibodyTree {
           "Joint '" + name + "' in model instance " +
           instance_index_to_name_.at(*model_instance) + " is not of type '" +
           NiceTypeName::Get<JointType<T>>() + "' but of type '" +
-          NiceTypeName::Get(GetJointByName(name)) + "'.");
+          NiceTypeName::Get(GetJointByName(name, instance)) + "'.");
     }
     return *typed_joint;
   }

```

I think?  But I haven't the time to test that up properly just now so I'm filing it instead.

sherm1, 2019-04-18:
Thanks, @ggould-tri! Assigned to @amcastro-tri to take a look.

amcastro-tri, 2019-04-18:
I'll push a fix soon. The analysis is correct @ggould-tri, thanks!. I'll extract the bit to ask for joint type in #11257 and use that (I also imagine this will merge faster).

PR #11280 (2019-04-19): GetJointByName: better error message for wrong joint type.
Resolves #11275.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11280)
closed 2019-04-19 by amcastro-tri