RobotLocomotion/drake #11287: PortSwitch primitive system

RussTedrake, 2019-04-19:
As discussed with a few of you, I've got the "PortSwitch" system mostly coded up now, and will PR it this weekend.  It's a simple system that offers many ports on the input, but passes-through only one of them to the output (with one additional input for the port_selector).  I imagine this being useful for combining many low-level controller primitives (e.g. DifferentialIK, PlanEval, hybrid control, etc) with all of them wired up, but only one of them getting called thanks to this switch.

cc @kmuhlrad @pangtao22

avalenzu, 2019-04-19:
This is the [Multiport Switch](https://www.mathworks.com/help/simulink/slref/multiportswitch.html) block in Simulink, in case you're interested in naming precedents (not that we need to be though).

pangtao22, 2019-04-19:
Thanks! Are the inputs to the PortSwitch system vector or abstract-valued?

RussTedrake, 2019-04-19:
It supports either.  Will PR soon.

PR #11290 (2019-04-30): Adds primitive PortSwitch system
Resolves #11287

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11290)
commit bb5e9f1d (2019-04-30): Adds primitive PortSwitch system

Resolves #11287
closed 2019-04-30 by RussTedrake