RobotLocomotion/drake #11291: Can't scalar convert between BasicVector types?

RussTedrake, 2019-04-20:
In #11290 (and in https://github.com/RobotLocomotion/drake/blob/master/systems/primitives/pass_through.h before it), I believe that we cannot setup the ports with a model_vector that will survive scalar conversion.  This is because I think we are missing a way to do scalar conversion between `BasicVector` types.  (e.g. if I have a `BasicVector<double>&` which is actually a `MyBasicVector<double>`, I can `Clone` to get another `MyBasicVector<double>`, but I don't know how to get a `MyBasicVector<AutoDiffXd>`.)

I know that we are increasingly feeling that ports using derived versions of `BasicVector` become awkward fast.  But if we do want to support them, then I think this piece is missing.

cc @jwnimmer-tri

jwnimmer-tri, 2019-04-23:
For PassThrough, I agree it would be awkward right now to try to offer a BasicVector-subclass constructor.  (It doesn't currently offer this, so at least it is merely impoverished, not broken.)  We _could_ do it by capturing the template type in the constructor overload and remembering the three constructor functors that we need.  In the case of PassThrough though, neither that nor new transmogrification operators on BasicVector are the right answer.  The right answer is that PassThrough's output port type should be inferred from its input port type, full stop.  Having the user provide a model value is a shim.

For PortSwitch, inferring the output type from the inputs would still be great, but perhaps impractical.  We might want to have a way to provide the model vector value but remember its three other constructors for later.

In any case, the easy work-around is to make a TypeCast primitive system that reapplies the vector subtype, then cascade that after any primitive that fails to express the correct runtime type on its output port.  That way, only one primitive system needs transmog hacks for BasicVector subtypes, instead of all of them.

RussTedrake, 2023-02-17:
I'm going to close this with the understanding that #12566 is the preferred solution going forward.

closed 2023-02-17 by RussTedrake