RobotLocomotion/drake #11294: urdf parser does not parse dynamics damping tag

RussTedrake, 2019-04-21:
Using http://underactuated.mit.edu/src/double_pendulum/double_pendulum.urdf from MBP results in different dynamics than RBP, because the MBP parser is not reading the `<dynamics damping="0.1" />` tags.

It seems like we certainly could:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_revolute_joint.html#aad91b55e6b714db4ba96190e6fab07ac

RussTedrake, 2019-04-21:
Update: The sdf parser seems to ignore the tag as well.

sammy-tri, 2019-04-22:
I'm not sure this is a parser problem.  Both the URDF and SDF parsers are tested to properly parse the damping in the acrobot model at https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/test/acrobot_parser_test.cc#L119  (I'd added an additional test to `detail_urdf_parser_test.cc` based on the double pendulum to verify before I noticed the existing test, so I'm pretty confident of the parsing).

Will ping the dynamics team to see if they've got any ideas.

sherm1, 2019-04-22:
Any thoughts on this, @amcastro-tri ?

RussTedrake, 2019-04-22:
i've managed to resolve it.  apologies.
(my urdf had a drake_ignore tag earlier that was cancelling the effect, and my sdf had the dynamics tag outside of the axis tag -- a perfect storm!).

closed 2019-04-22 by RussTedrake

RussTedrake, 2019-04-22:
*and* rigid body tree includes the damping forces in the "bias term", but MBP does not (it comes as a separate external force).  sigh...