RobotLocomotion/drake #11295: sdf parsing should support the `<gravity>` tag in `<world>`

RussTedrake, 2019-04-21:
We don't support this yet -- as documented here:
https://github.com/RobotLocomotion/drake/blob/master/multibody/parsing/detail_sdf_parser.h#L17-L18

But this means we have to add the ugly `AddForceElement` line (+ explicitly importing `UniformGravityFieldElement`) for every single example -- even the ones that should be trivially short.

E.g. 
```
from pydrake.all import MultibodyPlant, Parser, UniformGravityFieldElement
from underactuated import FindResource

plant = MultibodyPlant()
parser = Parser(plant)
parser.AddModelFromFile(FindResource("pendulum/pendulum.sdf"))
plant.AddForceElement(UniformGravityFieldElement())
plant.Finalize()
...
```

sammy-tri, 2019-04-23:
http://sdformat.org/spec?ver=1.6&elem=world#world_gravity implies to me that any model loaded from an SDF has a default gravity, which currently we're not adding.  It would be simple enough to do this (and I think for consistency this implies all models loaded from URDF also get default gravity).

MBP expects that a plant will have 0 or 1 gravity fields present and uses an overloaded implementation of `AddForceElement` to ensure this.  For compatibility with existing code and to simplify the parser implementation when loading multiple models, I'm thinking that we could make this overload idempotent so long as subsequent invocations specify a gravity field within epsilon of the existing field.

jwnimmer-tri, 2019-04-23:
It's long been on my wishlist (and already blessed by dynamics team) to have MBP start out with a default gravity field enabled, that the user would have to either replace with a new one, or else opt-out of by setting to nullptr or something.  That's probably the right place for this fix, instead of the parser.  And the "use weird template specialization junk" would disappear at the same time, making the parser's job easier for when there _is_ a gravity element also.

sammy-tri, 2019-04-23:
From an earlier slack conversation, it is also desirable to remove the automatically added `UniformGravityFieldElement` (or at least zero its magnitude)

RussTedrake, 2019-04-23:
Having a default gravity field is sufficient to resolve my current request/pain point.

sammy-tri, 2019-04-25:
If we get rid of the template specializations, that eliminates the part of my idea above where we have an idempotent version of the specialization...  which also means that existing code which we don't find and convert would end up with two gravity field elements (and presumably 2g of gravity).  Maybe that's OK since the process to fix is just to rip out (almost) every manual creation of a `UniformGravityFieldElement`.

jwnimmer-tri, 2019-04-25:
I'd say the specialization should remain for now, but marked as deprecated, and delegating to the new way, so that there is still only a single gravity.

jwnimmer-tri, 2024-03-16:
> Having a default gravity field is sufficient to resolve my current request/pain point.

In #11401, gravity is enabled by default.

If someone needs to change gravity via a custom SDFormat stanza, then we can re-open.  Until then, closing the parser request as "not planned".

closed 2024-03-16 by jwnimmer-tri