RobotLocomotion/drake #11303: install: Do not install Bazel-generated python scripts

EricCousineau-TRI, 2019-04-22:
Installing these scripts is worse than useless, it's harmful.

Lost time to this here: https://github.com/robotlocomotion/drake/pull/11261#issuecomment-485510181

jamiesnape, 2019-04-25:
They didn't used to be; we create different launchers, where necessary. Did something change?

EricCousineau-TRI, 2019-04-25:
Yeah, just that I had decided to install scripts under `pydrake.examples`, and didn't think to scrub the Bazel-generated stuff. #11304 seems simple enough to fix that mistake

jwnimmer-tri, 2020-05-02:
For the record, these appear to be the installed stubs, as of Drake v0.17.0.
```
drake/lib/python3.6/site-packages/pydrake/examples/multibody/pendulum_lqr_monte_carlo_analysis
drake/lib/python3.6/site-packages/pydrake/examples/multibody/cart_pole_passive_simulation
drake/lib/python3.6/site-packages/pydrake/examples/multibody/run_planar_scenegraph_visualizer
```

EricCousineau-TRI, 2020-05-02:
Just curious, can post your recipe used to generate that list?

jwnimmer-tri, 2020-05-02:
I grepped the v0.17.0 binary release for 'env python$'.

jwnimmer-tri, 2022-02-12:
The new grep is for `'env python3$'`.  As of v0.39.0, the list of problematic files remains only the same three as above.

PR #16583 (2022-02-14): [pydrake] Stop installing useless bazel py_binary stubs
Closes #11303.

This doesn't fail-fast in case a future developer ever tries to install a `py_binary` again, but _shrug_.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16583)
closed 2022-02-14 by rpoyner-tri