RobotLocomotion/drake #11315: Rename/deprecate MBP::get_continuous_state_output_port()

amcastro-tri, 2019-04-23:
## Problem

Currently the name of this method is a lie. This port outputs the state of the plant whether it is continuous or discrete. 

## Proposal
- Refactor `MBP::get_continuous_state_output_port()` to `MBP::get_state_output_port()`.
- Deprecate `MBP::get_continuous_state_output_port()`.

cc'ing @avalenzu, @jwnimmer-tri

sherm1, 2019-04-24:
Please hold off deprecating this until we decide on what names to use for the repaired state abstractions in #9171. One thing I've been considering is to redefine "continuous" to mean "differentiable" in which case the current output port name would be OK. (Whether these variables get updated by an integrator or a discrete event handler would not change their name.)

avalenzu, 2019-04-24:
> One thing I've been considering is to redefine "continuous" to mean "differentiable" ...

That'll have interesting repercussions for large swaths of mathematics . . .

sherm1, 2019-04-24:
> That'll have interesting repercussions for large swaths of mathematics

Yeah, could be a very bad idea! I'm looking for a way to fix our variable abstractions that doesn't do too much violence to our existing API, which leans heavily on DeclareContinuousState() to create q and v. Whether those get updated by an integrator or event handler or both is a separate matter. This is likely a good topic for an interested-parties discussion at the onsite.

amcastro-tri, 2019-04-24:
Whatever the system's abstractions churn looks like, IMO we should aim for having just `MBP::get_state_output_port().`, it conveys what everyone has in mind when talking about state, no?

sherm1, 2019-04-24:
Yes, you're right -- that's better regardless of how the state naming comes out.

SeanCurtis-TRI, 2019-04-24:
My one counter thought is that the System framework has the concept of "state" that includes q, v, z, abstract, and discrete. In this context, we'd argue that abstract and discrete aren't included. But relying on context to take a system-wide concept and change it locally seems dangerous.

PR #11323 (2019-04-25): Deprecate MBP::get_continuous_state_output_port()
Resolves #11315.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11323)
closed 2019-04-25 by amcastro-tri