RobotLocomotion/drake #11317: Doxygen for MathematicalProgram is unreadable

ggould-tri, 2019-04-24:
Doxygen rendering of return types makes `drake::solvers::MathematicalProgram` documentation nearly unreadable.

Visit https://drake.mit.edu/doxygen_cxx/classdrake_1_1solvers_1_1_mathematical_program.html
Observe that the page doesn't contain any content because everything is off the right margin:

[image]

Note that the layout culprit is the return type of `GetInitialGuess`, which is `std::enable_if< std::is_same< typename Derived::Scalar, symbolic::Variable >::value, Eigen::Matrix< double, Derived::RowsAtCompileTime, Derived::ColsAtCompileTime > >::type` (that's a mouthful!).

The solution might be either to give doxygen a better custom CSS or else to typedef some of the more awful return type subclauses.

jwnimmer-tri, 2019-04-24:
The solution that comes to my mind is to put ifdef doxygen around the return type, as I did in https://github.com/RobotLocomotion/drake/blob/2b41c1b655acd2fa86902b84fc53e8b9857cd6a2/systems/framework/input_port.h#L68.  It's probably going to be difficult to get typedefs to work.

j4yyousi, 2025-11-18:
Hi, here are the before and after pictures showing the effect of removing `std::enable_if` and adding `requires` in `/doxygen_cxx/classdrake_1_1solvers_1_1_mathematical_program.html`:
Before:
[image]

After:
[image]

The result is that function signatures get shifted to the left. However, `std::tuple` return types (like `std::tuple< symbolic::Polynomial, MatrixXDecisionVariable, MatrixXDecisionVariable>` ) are mostly long which prevents function signatures from shifting more to the left. Should we alias the tuples or the result in the _after_ picture is satisfactory?

jwnimmer-tri, 2025-11-18:
Well, "after" is certainly superior to "before", so we could accept that as as PR with nice progress.

For the tuples, I wonder if using `auto` as the return type would make things show up better?  And then explain the details of the `tuple` either as a ["trailing return type"](https://en.wikipedia.org/wiki/Trailing_return_type) (if Doxygen happens to show that better than a leading one), or just leave it as just "auto" and explain the details of the tuple only in the API documentation prose, not the function signature.