RobotLocomotion/drake #11318: Weird symbolic differentiation

sadraddini, 2019-04-24:
I'm running into some symbolic problems. Consider differentiating `sin(x)*cos(x)` with respect to `x`.
```
> import pydrake.symbolic as sym 
> x=sym.Variable("x")
> e=sym.sin(x)*sym.cos(x)
> e.Differentiate(x)
```
What I will get is
```
<Expression "(((pow(sin(x), -1) * cos(x)) - (sin(x) * pow(cos(x), -1))) * sin(x) * cos(x))">
```
Of course this is correct, but not the best representation. If I want to evaluate `e` at `x=0`, I will get `NaN`.
```
> e.Differentiate(x).Evaluate({x:0})
nan
```
Of course I can use `pydrake.symbolic.Expression.Expand` to first simplify and then evaluate - it gives the correct answer:
```
> e.Differentiate(x).Expand().Evaluate({x:0})
1.0
```
However, the problem with using Expand is i) it slows down symbolic computations, and ii) currently I can not find a vectorized Expand function - I wish there was an expand or simplify (#8515) operation for numpy arrays. I do not know the engine behind symbolic differentiation, but is it fixable?

soonho-tri, 2019-04-24:
@sadraddini , thanks for the report. I'll check it.

soonho-tri, 2019-04-24:
@sadraddini , can you check the latest version of drake and try the example again? I have:

```
<Expression "( - pow(sin(x), 2) + pow(cos(x), 2))">
```

soonho-tri, 2019-04-24:
FYI, I think the behavior was changed/fixed by https://github.com/RobotLocomotion/drake/pull/11099.

soonho-tri, 2019-04-24:
> ii) currently I can not find a vectorized Expand function

FYI, you can define one:

```py
x=sym.Variable("x")
y=sym.Variable("y")
expand = np.vectorize(lambda e: e.Expand())    # <---

l = np.array([(x + y)**2, (x + y)**3])
print(expand(l))
```
output:
```
[<Expression "(2 * (x * y) + pow(x, 2) + pow(y, 2))">
 <Expression "(3 * (x * pow(y, 2)) + 3 * (pow(x, 2) * y) + pow(x, 3) + pow(y, 3))">]
```

sadraddini, 2019-04-24:
Thanks! My latest version of Drake was built just days before fixing #11099. Problem resolved. Regarding ii) using python `lambda` is not efficient (I have [heard](https://www.quora.com/Is-it-better-to-use-list-comprehensions-or-map-with-a-lambda-function-when-manipulating-elements-in-a-list) it can be only as good as running a for loop). Anyways thank you the first part. I will pull the latest drake.

closed 2019-04-24 by sadraddini

soonho-tri, 2019-04-24:
> Regarding ii) using python lambda is not efficient (I have heard it can be only as good as running a for loop).

From https://docs.scipy.org/doc/numpy/reference/generated/numpy.vectorize.html :

> The vectorize function is provided primarily for convenience, not for performance. The implementation is essentially a for loop.

So, I think you're right. In the end, it's just a for loop. But as noted above, `np.vectorize` is designed for convenience and readability, not for performance.