RobotLocomotion/drake #11319: Doxygen link to "@ref tangent_force" is dead

amcastro-tri, 2019-04-24:
A user just found out that [this link](https://github.com/RobotLocomotion/drake/blob/9c40c8c0d10cf2a784b39f2eeb98478bb00a5cf8/multibody/plant/multibody_plant.h#L2879) is dead and we'd like to bring back the nice description we had about the Stribeck model (including a great schematic!).

cc'ing @SeanCurtis-TRI who is the original author of those nice docs.

SeanCurtis-TRI, 2019-04-24:
It was removed in #10622. It probably makes sense to resurrect the parts that are actually still relevant to the MBP model in the MBP documentation.

amcastro-tri, 2019-05-09:
I bumped the priority of this because just in the last month I had several requests for this, including external people from other companies.

SeanCurtis-TRI, 2019-05-13:
Currently under review.

PR #11431 (2019-05-15): Contact modeling docs.
Resolves #11319.
Resolves #11470.

Brings back the awesome pieces of docs written by @SeanCurtis-TRI. 
New section on the physics of contact has been added and older sections were brought up-to-date with our current code. Removed no longer relevant sections.

Priority is high given the high demand for this from users.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11431)
closed 2019-05-15 by amcastro-tri