RobotLocomotion/drake #11321: Easier way to get poses from a Python PoseBundle object

kmuhlrad, 2019-04-24:
It'd be nice to have a better way to get a specific pose out of a `PoseBundle` object. Whenever I work with `PoseBundle`s, I almost always write something similar to this, so having something similar to it as a class method would be nice.

```
pose_dict = {}

for i in range(pose_bundle.get_num_poses()):
    if pose_bundle.get_name(i):
        pose_dict[pose_bundle.get_name(i)] = pose_bundle.get_pose(i)
```

EricCousineau-TRI, 2019-04-24:
Per Slack convo, this is for named accessors. This should be done in C++ to maintain API parity, if it is to be added to Drake API.

jwnimmer-tri, 2019-04-25:
To @sherm1 for delegation.

FYI I interpret Eric's "named accessors" to mean "Add a method to retrieve a pose by its string name".

kmuhlrad, 2019-04-25:
I guess adding a method to retrieve poses by model ID would also be helpful, but I've personally never done that and only use their names.

sherm1, 2019-04-25:
One complication with this is that the names are not required to be unique. Only a `(name, model_id)` pair is guaranteed unique. Elsewhere (MBP, e.g.) we have allowed name-only lookup but only if the name is unique across all model ids. I'm not sure what the implications would be for making this nice in Python.

cc @SeanCurtis-TRI

EricCousineau-TRI, 2019-04-25:
Ah, did not know that this also has the mystical coupling between multibody information via `model_instance`... blech...

Regarding Python, nothing is difficult there; it should be exactly the same as C++.

jwnimmer-tri, 2019-04-25:
TBH PoseBundle is kind of old and smelly, coming from before the time we had SceneGraph and its view of the world.  Once Drake automotive leaves the tree, the _only_ in-tree use of PoseBundle will be for PoseBundleToDrawMessage, and _that_ System is kind of a bad idea in the first place.  I think we could (and should) rework the drake-visualizer interactions, at which point PoseBundle goes "poof" anyway.  I'm not sure how much we should be adding features to doomed classes.

kmuhlrad, 2019-04-25:
What are some alternatives to PoseBundle for passing around multiple poses?

jwnimmer-tri, 2019-04-25:
If SceneGraph is in play, then [FramePoseVector](https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1geometry.html#aabe48813dfd7db14c564904dcefab397) is the way to go (a map from `geometry::FrameId` to poses.

Otherwise, you can use your own abstraction.  If it's all pydrake, for example, a `dict` from `str -> Isometry3d` or `str -> RigidTransformd` sounds like what you want (if you're only using the pose names anyway).

sherm1, 2019-04-25:
My 2 cents written before I saw @jwnimmer-tri's above:

Looking at the code and consulting with @SeanCurtis-TRI, it appears that PoseBundle is to be phased out at some point. There are TODOs there to that effect. The class doesn't contain fast-access data structures for dealing with names and model instances. I don't think we should put effort into upgrading that class, and I wouldn't want to hide the expensive lookups. The current interface makes it clear that access is slow, and as a result @kmuhlrad appropriately built a map to make subsequent lookup fast. That's good!

My proposal would be to leave this alone.

EricCousineau-TRI, 2019-04-25:
That sounds good to me.

@kmuhlrad In this case, I'd suggest making a utility file for the sugar you've already done, so you can reuse your pattern for the time being. If you'd like some suggestions, please let me know.

Once that's done, I'd propose we close this issue; if there's add'l sugar for `FramePoseVector`, then we'd prolly want to open a new issue.

kmuhlrad, 2019-04-25:
Do you want me to PR that file?

EricCousineau-TRI, 2019-04-25:
Ah, sorry - I meant that you keep the utility local to your app code. I'd say we keep it out of Drake for now, and defer any sugar to after `PoseBundle` evaporates.

tri-ltyyu, 2019-04-26:
Closing as won't fix in Drake (fix downstream)

closed 2019-04-26 by tri-ltyyu