RobotLocomotion/drake #11326: Allow symbolic::Polynomial::DoAddProduct not to call `Expand`

hongkai-dai, 2019-04-25:
In `symbolic::Polynomial::DoAddProduct`, it takes a lot of time in this line
https://github.com/RobotLocomotion/drake/blob/564935e9ef45be9f8378674289398c515ae54bd1/common/symbolic_polynomial.cc#L45
when it calls `Expand`.

In one of my optimization program, it takes more time to compute the polynomial than the Mosek solver to solve the polynomial optimization problem, as shown in this callgrind output
[image]
The actual `solvers::Solve` takes only 20% of time, while `Expand` takes 60%.

I would argue for some cases, it makes sense not to check if we can simplify the polynomial addition `p1 + c * p2`. For example, when I add the constraint that the coefficient of the polynomial `p1 + c * p2` should all be 0, we called `Expand` when adding the linear equality constraint through MathematicalProgram::AddLinearEqualityConstraint, in which we call `DecomposeLinearExpression`.

A potential solution is to add an optional argument `simplify_flag` to `AddProduct` as
```C++
Polynomial& Polynomial::AddProduct(const Expression& coeff, const Monomial& m, bool simplify_flag = true);
```
when the flag is set to true, we check if we can remove some monomial-coefficient pair by calling `Expand` function; otherwise we skip this check and add the coefficients directly.

cc @soonho-tri

soonho-tri, 2019-04-25:
SGTM. I'll extend the signature.

jwnimmer-tri, 2022-05-01:
@hongkai-dai I wonder if our recent changes have sufficiently fixed this problem?

hongkai-dai, 2022-05-01:
I will test it on a larger sum-of-squares program on my dev branch and see how much speed up we obtain. I will report my findings later.