RobotLocomotion/drake #11341: Convert Drake internal uses of context.FixInputPort() to port.FixValue()

sherm1, 2019-04-26:
PR #10759 added `port.FixValue()` as a nicer API than `context.FixInputPort(port_index)`. PR #10964 converted some Drake internal uses to FixValue(). There are still lots of the old style uses in place that should be converted.

Please fix these as you run into them, especially in examples where users might see them.

mpetersen94, 2020-01-16:
Here is the edited list of the remaining files that still include `context.FixInputPort(port_index)`
```
bindings/pydrake/examples/multibody/run_planar_scenegraph_visualizer.py
bindings/pydrake/systems/framework_py_semantics.cc
bindings/pydrake/systems/test/general_test.py
systems/framework/*
systems/primitives/test/affine_linear_test.h
tutorials/dynamical_systems.ipynb
```

sherm1, 2020-01-16:
That list is getting very short -- thanks, Mark!

sherm1, 2020-09-14:
That's a good question. port.FixValue() is implemented via the context_base.FixInputPort(port, AbstractValue) method so that has to survive (though could be hidden from doxygen or even made private with a loophole for InputPort access). Could be worthwhile to deprecate the `Context<T>` sugar signatures of FixInputPort -- FixValue handles the sugar better. (We could just leave the public ContextBase method -- that signature isn't used much and we can leave a note there recommending the port method.)

Anyone have opinions about keeping or removing context.FixInputPort() methods?

mpetersen94, 2020-09-16:
Would a slack poll be more effective at soliciting public comment? I don't know how many people are tracking this issue but many downstream users seem to use this workflow.

sherm1, 2020-09-16:
Worth a try -- do you want to try one? DrakeDevelopers #general channel would be a good spot I think.

sherm1, 2020-10-02:
Done -- thanks, @mpetersen94!

closed 2020-10-02 by sherm1