RobotLocomotion/drake #11346: Add more safe Set() methods to Context for state modification

sherm1, 2019-04-27:
With caching enabled we want to discourage users from holding mutable references to state variables. The cache is invalidated (once) when mutable access is granted, but subsequent writes through the reference can't cause more invalidation.

Add some more safe Set...State() methods to Context to encourage good state hygiene.

#### Motivation

Variants of this are common in existing code:
```c++
systems::VectorBase<double>& xc = 
    context.get_mutable_continuous_state_vector();
xc[0] = x0;
xc[1] = v0;
```
There is nothing wrong with the code above as written, but is a dangerous practice because of the temptation to perform a computation like this:
```c++
auto& xc = context.get_mutable_continuous_state_vector();
xc[0] = x0;
xc[1] = v0;
// Do some computation.
xc[0] = new_x_value;
// Repeat computation with new value. <-- FAIL
```
That second computation won't happen since the caching system will think it's up to date.

Something like the following would encourage safe coding:
```c++
context.SetContinuousState(0, {x0, v0});
// Do some computation.
context.SetContinuousState(0, new_value);
// Repeat computation with new value. <-- WIN
```
#### Tasks
- [ ] Design and implement enough APIs to clean up most use cases.
- [ ] Don't forget about discrete and abstract state variables.
- [ ] Modify existing code, especially examples, to use the safer methods.
- [ ] Consider if there is any hope for generated BasicVector subtypes used as states - are these inherently unsafe?
- [ ] Make sure the new APIs are consistent with upcoming #9171 state variable changes.

RussTedrake, 2021-07-31:
@sherm1 -- this was marked as high priority (I think correctly), and I know at least `SetContinuousState` and `SetDiscreteState` methods exist.  Can you update this issue?  I'm considering some usability improvements along these lines, too.

sherm1, 2021-08-02:
Update: a quick inventory as of 8/2/2021.

The following safe and convenient `context.Set...()` methods currently exist:
```
SetAbstractState<V>()
SetAccuracy()
SetContinuousState()
SetDiscreteState()
SetTime()
SetTimeAndContinuousState()
```
These take scalars and nice Eigen vectors, no BasicVectors or other fluff. These should be used instead of the dangerous and awkward `get_mutable_...()` methods.

Missing from that list:
```
SetAbstractParameter<V>()
SetNumericParameter()
```

The following dangerous cache-invalidating methods are improperly using snake_case names. They should be deprecated and replaced with CamelCase methods and marked "(Advanced)":
```
get_mutable_abstract_parameter() -> AbstractValue&
get_mutable_abstract_state() -> AbstractValues&
get_mutable_abstract_state<U>(i) -> U&
get_mutable_continuous_state() -> ContinuousState&
get_mutable_continuous_state_vector() -> VectorBase&
get_mutable_discrete_state() -> DiscreteValues&
get_mutable_discrete_state_vector(i) -> BasicVector&
get_mutable_numeric_parameter(i) -> BasicVector&
get_mutable_parameters() -> Parameters&
get_mutable_state() -> State&
```
All Drake code that uses these unnecessarily should be converted to use the `Set...()` methods instead.

RussTedrake, 2021-08-02:
Will you add the corresponding `Get...()` methods, too?  (pretty please?)

sherm1, 2021-08-03:
Do you mean the non-mutable methods with copied-out Eigen return values?
```c++
GetContinuousState() -> VectorX  // copy of qvz packed together (proposed?)
get_continuous_state() -> const ContinuousState& // already exists
get_continuous_state_vector() -> const VectorBase& // already exists
```
what else to you have in mind? For discrete state, I don't understand why the existing `get_discrete_state_vector()` returns a `const BasicVector&` when it could have returned a `const VectorX&`. Should we make a somewhat-painful change to the return type? Or maybe add `get_discrete_state_value() -> const VectorX&` and deprecate the other method?

sherm1, 2021-08-03:
@rpoyner-tri any thoughts on what we should do to make state access more pleasant?

RussTedrake, 2021-08-04:
You've captured the request correctly.  Something like `Context::GetContinuousState()` and `Context::get_discrete_state_value()` (with 0 and 1 argument) is the ask.  As you point out, the only subtlety is picking the names well.

After this, I would *suspect* that the vast majority of users would access the `Context` members via:
```
context.SetContinuousState(vec)
vec = context.GetContinuousState()

context.SetDiscreteState(vec)
context.SetDiscreteState(index, vec)
vec = context.get_discrete_state_value()
vec = context.get_discrete_state_value(index)

context.SetTime(t)
t = context.get_time()
...
```

That is the immediate ask.  Please don't let the remainder distract from that easy(?) fix.

**Additional Idea 1:**
It is tempting to offer `context.GetDiscreteState()` and `context.GetDiscreteState(index)` (perhaps in addition to the lower case variants)... even though the CamelCase overstates the cost of calling the method, the value here of presenting a consistent interface to the user would be immense.  The small percentage of users who understand that the lower case means cheap can call the lowercase version.  Our docstrings for `GetDiscreteState` would, of course, document that fact.

That would give the majority of users:
```
context.SetContinuousState(vec)
vec = context.GetContinuousState()

context.SetDiscreteState(vec)
context.SetDiscreteState(index, vec)
vec = context.GetDiscreteState()
vec = context.GetDiscreteState(index)
```
But then do we go further down this rabbit hole, offered `context.GetTime()`, etc...? Especially since want to be moving CamelCase method to lowercase methods by making them more efficient (e.g. `ContinuousState` hasa `BasicVector`)

**Additional Idea 2:**
To state the obvious: the fundamental reason for the set/get asymmetry is that we can overload the method name for arguments but not return values.  `SetContinuousState(ContinuousState&)` and `SetContinuousState(VectorX&)` can co-exist (as can `set_continuous_state(ContinuousState&)` and `set_continuous_state(VectorX&)`), but we have to pick one type for the return.  I do think it's bad form to have `GetDiscreteState()` and `get_discrete_state()` co-exist just so they can return different types.

So an alternative might be to say we should not have offered `SetContinuousState` as is.  Instead we should have provided:
```
context.SetContinuousStateValue(vec)
vec = context.GetContinuousStateValue()

context.SetDiscreteStateValue(vec)
context.SetDiscreteStateValue(index, vec)
vec = context.GetDiscreteStateValue()
vec = context.GetDiscreteStateValue(index)
```
(or `Value=>Vector` if, as you say, we go through the deprecation).  The cost of writing any one call goes up by a few characters, which isn't ideal.  But the value of consistency might trump the value of having a shorter name.  I even think that
```
context.SetContinuousStateValue(vec)
vec = context.GetContinuousStateValue()

context.SetDiscreteStateValue(vec)
context.SetDiscreteStateValue(index, vec)
vec = context.get_discrete_state_value()
vec = context.get_discrete_state_value(index)
```
is slightly better than the proposal at the top of this comment.  (btw -- `SetDiscreteStateValue` is really `CamelCase`?)