RobotLocomotion/drake #11349: Drake's binary release of libbot2 depends on outdated openjdk-8-jre

jwnimmer-tri, 2019-04-27:
Drake's PPA build of `libbot2` declares `Depends: openjdk-8-jre`, but I think that is too specific.  On Bionic, the `default-jre` is now `openjdk-11-jre`.

On TRI's system `libbot2` is the _only_ thing that brings in JRE 8, and having two JREs installed is undesirable.

I think the correct `Depends` item is more like `default-jre | java8-runtime`.

jamiesnape, 2019-04-29:
Depends if it works with both JDKs, which is not a given. I will look into it.

jamiesnape, 2019-07-22:
Apt repo has been updated with lcm 1.4.0 and libbot2 0.0.1.20190722.

jwnimmer-tri, 2019-07-22:
I am not seeing them in the index:
```console
(bionic)jwnimmer@call-cc:~$ cat /etc/apt/sources.list.d/drake.list 
deb [arch=amd64] https://drake-apt.csail.mit.edu/bionic bionic main

(bionic)jwnimmer@call-cc:~$ sudo apt -q -q update
All packages are up to date.

(bionic)jwnimmer@call-cc:~$ sudo apt install lcm libbot2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
lcm is already the newest version (1.3.95.20180523-1).
libbot2 is already the newest version (0.0.1.20180523-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

jamiesnape, 2019-07-22:
Maybe it hadn't finished propagating to all the cloudfront endpoints. I have
```
$ apt download --print-uris lcm libbot2
'https://drake-apt.csail.mit.edu/bionic/pool/main/l/lcm/lcm_1.4.0-1_amd64.deb' lcm_1.4.0-1_amd64.deb 2468816 SHA512:34bd5098f6d26482c916e8392a4a890fd8d8926e08f2665486b2d5394eb91e40f876f386266cfe9d46466a4391c5159a2fcd3b4d18195c9d6b747897d9ab95ed
'https://drake-apt.csail.mit.edu/bionic/pool/main/libb/libbot2/libbot2_0.0.1.20190722-1_amd64.deb' libbot2_0.0.1.20190722-1_amd64.deb 994458 SHA512:0c2647dae5b6c61aae19046cf2f94579d363420cc41689b95ff10472dfbe34cc188be7e9106f8b4b2650fd6e3c3d3461da9fd5817fe0d6634c51ae2ba10f44da
```

jwnimmer-tri, 2019-07-22:
Indeed, I see them now.

jwnimmer-tri, 2019-08-15:
The new packages are indeed on the site now, and have the fixed Depends line.

closed 2019-08-15 by jwnimmer-tri