RobotLocomotion/drake #11363: pydrake attic shims fail to generate warnings

jwnimmer-tri, 2019-05-01:
In #11358, I removed the attic shims for pydrake import paths, but then in #11361 we put them back, because `underactuated` builds were broken and apparently had not been triggering the deprecation warnings.  Probably users other than underactuated could have experienced the same breaks.

We should figure out if indeed the warnings were not triggering, and then try to make them trigger again.  Maybe even push back the date beyond 2019-06-01.

EricCousineau-TRI, 2019-05-02:
FTR Comment indicating failures in `underactuated`: https://github.com/RobotLocomotion/drake/pull/11358#pullrequestreview-232593089

EricCousineau-TRI, 2019-05-23:
I need to follow up on this next week.

EricCousineau-TRI, 2019-06-06:
Given that I haven't done this, we should definitely push back the removal date.

@jwnimmer-tri Should we push it back by 3 months? :(

jwnimmer-tri, 2019-06-06:
My default suggestion is that the date should be 3 months after whenever the warnings start working.

EricCousineau-TRI, 2019-06-13:
Starting work on this now; sorry for the delay!

EricCousineau-TRI, 2019-06-13:
Initial guess is that for `meshcat_rigid_body_visualizer.py`, the use of `pydrake.all` may have suppressed the warning, possibly because I screwed up the `stacklevel` in the deprecation warning:
https://github.com/RussTedrake/underactuated/blob/89ecc94de9de1ecd8c9335335a986ec6935618e0/src/underactuated/meshcat_rigid_body_visualizer.py#L28
https://github.com/RobotLocomotion/drake/blob/12d3158a56feaa178432977dc6b89073155cd4bc/tools/skylark/alias.bzl#L134-L145

\cc @RussTedrake

EricCousineau-TRI, 2019-06-13:
Hmm... While it's true that I screwed up `stacklevel`, the fact that `pydrake.all` is used at all means that it will always swallow those warnings.

It isn't fixable by anything simple. Basically, importing `pydrake.all` includes all modules, including those that are deprecated. Because the deprecated modules are loaded, the deprecation statement will *not* retrigger upon subsequent imports (like in `underactuated.multibody`) since the module is loaded; setting `simplefilter("always", ...)` won't affect anything, because the module is already loaded.

I'll submit a PR that documents the above stuff - that having `pydrake.all` in tested code will swallow module deprecations, and in a way that is nigh impossible to detect; existing docs only touch on the fact that deprecated symbols may just flat-out disappear:
https://github.com/RobotLocomotion/drake/blob/12d3158a56feaa178432977dc6b89073155cd4bc/bindings/pydrake/all.py#L9-L13

The fix for `underactuated` is to not use `pydrake.all` (or any `*.all` module) in any tested code; in Jupyter notebooks or in class examples, I guess that'd be fine, *but*, it should always happen after all other code, with the understanding that it won't catch deprecation.

The symbols could be wrapped with deprecation, but that's additional overhead that I'd prefer we not spend any time on; we have enough shims as-is, and wrapping the imported classes naively will break stuff like `isinstance(x, RigidBodyTree)`; that's fixable-isih, but more engineering effort (that could maybe be better placed elsewhere?).

@RussTedrake Can I ask if you're OK with the simpler solution, which is just documenting that using `pydrake.all` has sharp edges for detecting deprecation?

EricCousineau-TRI, 2019-06-13:
Submitted #11645 - can discuss there

EricCousineau-TRI, 2019-06-20:
Solution path changed, see: https://github.com/RobotLocomotion/drake/pull/11645#issuecomment-504047804

PR #11645 (2019-06-25): pydrake: Remove catching of deprecation errors in `.all` modules
Resolves #11363

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11645)
closed 2019-06-25 by EricCousineau-TRI