RobotLocomotion/drake #11368: installation:Drake Source installation (macOS)

54WC, 2019-05-02:
I follow Drake Documentation » Installation and Quickstart » Source installation (macOS)
It look like no issue to run " ./setup/Mac/install_prereqs.sh" for MAC. 
In the "bazel build .... ", some warning appear, but finally build completed successfully. 
Next, I run "bazel-bin/examples/double_pendulum/double_pendulum_demo", runtime error happen.  The detail message is post below.  

1. MacBook-Pro:double_pendulum wc$ bazel build ...

Starting local Bazel server and connecting to it...
INFO: Analysed 35 targets (266 packages loaded, 26121 targets configured).
INFO: Found 35 targets...
INFO: From Linking common/libcond.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/common/libcond.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking math/libeigen_sparse_triplet.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/math/libeigen_sparse_triplet.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking math/libgradient.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/math/libgradient.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking math/libmatrix_util.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/math/libmatrix_util.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking common/libpointer_cast.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/common/libpointer_cast.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking math/liborthonormal_basis.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/math/liborthonormal_basis.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking math/libvector3_util.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/math/libvector3_util.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking common/libdouble.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/common/libdouble.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking math/libautodiff.a:
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: bazel-out/darwin-opt/bin/math/libautodiff.a the table of contents is empty (no object file members in the library define global symbols)
INFO: Elapsed time: 838.095s, Critical Path: 93.77s
INFO: 785 processes: 785 darwin-sandbox.
INFO: Build completed successfully, 1150 total actions

2. MacBook-Pro:drake$ bazel-bin/examples/double_pendulum/double_pendulum_demo

libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Could not find Drake resource_path 'drake/examples/double_pendulum/models/double_pendulum.sdf' because no resource roots of any kind could be found: DRAKE_RESOURCE_ROOT is unset, a bazel::tools::cpp::runfiles::Runfiles could not be created, and there is no Drake CMake install marker.
Abort trap: 6

jwnimmer-tri, 2019-05-02:
Please add `--spdlog_level=trace` to the command (to get more detailed diagnostics), and share that new output in this issue.

```console
$ bazel-bin/examples/double_pendulum/double_pendulum_demo --spdlog_level=trace
```

jwnimmer-tri, 2019-05-02:
Actually, I can reproduce the problem on a mac as well:
```
macsim:drake jwnimmer$ bazel-bin/examples/double_pendulum/double_pendulum_demo --spdlog_level=debug
[2019-05-02 11:32:51.177] [console] [debug] FindResource ignoring DRAKE_RESOURCE_ROOT because it is not set.
[2019-05-02 11:32:51.178] [console] [debug] _NSGetExecutablePath = /Users/jwnimmer/jwnimmer-tri/drake/bazel-bin/examples/double_pendulum/double_pendulum_demo
[2019-05-02 11:32:51.178] [console] [debug] FindRunfile mechanism = argv0
[2019-05-02 11:32:51.178] [console] [debug] FindResource ignoring CMake install candidate '...' because it does not exist
libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Could not find Drake resource_path 'drake/examples/double_pendulum/models/double_pendulum.sdf' because no resource roots of any kind could be found: DRAKE_RESOURCE_ROOT is unset, a bazel::tools::cpp::runfiles::Runfiles could not be created, and there is no Drake CMake install marker.
```

jwnimmer-tri, 2019-05-02:
Weird, the error from bazel Runfiles object is thus ...

```
[2019-05-02 11:36:24.806] [console] [debug] FindRunfile error: ERROR: external/bazel_tools/tools/cpp/runfiles/runfiles.cc(211): bad runfiles manifest entry in "/Users/jwnimmer/jwnimmer-tri/drake/bazel-bin/examples/double_pendulum/double_pendulum_demo.runfiles/MANIFEST" line #1: "�����(����H__PAGEZEROx__TEXT@�@�__text__TEXT0Z��u0Z�__stubs__TEXT�!vB�!v�__stub_helper__TEXT:v�" (created using argv0 with TEST_SRCDIR=nullptr and RUNFILES_MANIFEST_FILE=nullptr and RUNFILES_DIR=nullptr)
```

... but `/Users/jwnimmer/jwnimmer-tri/drake/bazel-bin/examples/double_pendulum/double_pendulum_demo.runfiles/MANIFEST` is the regular text file that I'd expect, not that binary line.

jwnimmer-tri, 2019-05-02:
I found the problem.  Our computed `argv0` has 38 nil bytes in it.  Hopefully I can fix it soon.

PR #11369 (2019-05-02): runfiles: Fix argv0 handling on macOS
Closes #11368.  Broken in #11261.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11369)
closed 2019-05-02 by jwnimmer-tri