RobotLocomotion/drake #11373: After Source installation Drake on macOS , what do I expect ?

54WC, 2019-05-03:
I follow Drake Documentation » Installation and Quickstart » Source installation (macOS). 
After bug 11368 is fixed, it looks like no bug. But, should I see some thing happen?  
For example, in case running "bazel-bin/examples/acrobot/run_swing_up_test", should I see visualize result ?  But like below result, it end up with nothing. 
Could you offer a book  about "Drake source installation for dummy" ?   

(1) case 1
Weide-MacBook-Pro:drake wc$ bazel-bin/examples/acrobot/run_swing_up_test
Weide-MacBook-Pro:drake wc$ bazel-bin/examples/acrobot/run_swing_up_test

(2) case 2
Weide-MacBook-Pro:drake wc$ bazel-bin/examples/van_der_pol/van_der_pol_test
Using drake_cc_googletest_main.cc
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from VanDerPolTest
[ RUN      ] VanDerPolTest.ScalarConversionTest
[       OK ] VanDerPolTest.ScalarConversionTest (6 ms)
[ RUN      ] VanDerPolTest.LimitCycle
[       OK ] VanDerPolTest.LimitCycle (7 ms)
[----------] 2 tests from VanDerPolTest (13 ms total)
[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (13 ms total)
[  PASSED  ] 2 tests.

jwnimmer-tri, 2019-05-03:
Sorry for the confusion / lack of documentation.

Most examples do not open a visualization window by default.  Before launching the example, launch the visualizer first:
```console
bazel-bin/tools/drake_visualizer &
bazel-bin/examples/acrobot/run_swing_up
```

The double_pendulum docs might help:
https://github.com/RobotLocomotion/drake/tree/master/examples/double_pendulum#how-do-i-run-it

54WC, 2019-05-03:
I invoke bazel to run three version of command for drake_visualizer, all result in the same bug:ImportError: No module named vtk.  In this case, I can't append "--spdlog_level=trace" to get more information. 

(1)MacBook-Pro:drake wc$ ./bazel-bin/tools/drake_visualizer
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
ImportError: No module named vtk

(2)MacBook-Pro:drake wc$ bazel-bin/tools/drake_visualizer 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
ImportError: No module named vtk

(3) MacBook-Pro:drake wc$ bazel run //tools:drake_visualizer 
INFO: Analysed target //tools:drake_visualizer (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //tools:drake_visualizer up-to-date:
  bazel-bin/tools/drake_visualizer
INFO: Elapsed time: 0.599s, Critical Path: 0.02s
INFO: 0 processes.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
ImportError: No module named vtk

closed 2019-05-03 by jwnimmer-tri